mavlink/MAVSDK #2422: what should I do if I want to apply some changes to plugins like action?

Steven-Ajex, 2024-10-10:
I'v add `_system_impl->set_flight_mode(FlightMode::Posctl); ` to arm() function to test if i can apply changes to plugin implications,while hoping that when arm() is called MAVSDK can send messages to switch the UAV's flight mode to position.
So,what should i do to build the MAVSDK to apply my changes?I followed the guidance in contribution to and ran auto-generation and building library from source but it did work,have i missed anything?I am grateful for any guidance and help.

julianoes, 2024-10-10:
This goes back to the feature request for a "set mode" API, I'd say.

Also see: #2128

Steven-Ajex, 2024-10-11:
> (You can always do any changes and build yourself but I usually don't recommend that as you are then on a fork that you need to maintain yourself.)

thanks,i'll be careful,there will not be a large difference,just some adaption to my self-built autopilot

Steven-Ajex, 2024-10-11:
> (You can always do any changes and build yourself but I usually don't recommend that as you are then on a fork that you need to maintain yourself.)

I'v solved to problem thanks a lot.

closed 2024-10-11 by julianoes