mavlink/MAVSDK #2426: ArduPilot SITL : SIGSEGV Sending message 76

depauwjimmy, 2024-10-16:
Hi,

I am trying the SDK again starting a new project.
I have a simulation environment running gazebo/ardu sitl/ros2 and i have an issue when creating System.

Ardu version : Copter-4.5.7
MAVSDK : 2.12.10

Running logs are

```
[03:15:53|Info ] MAVSDK version: v2.12.10 (mavsdk_impl.cpp:26)
[03:15:53|Info ] New system on: 127.0.0.1:33815 (with system ID: 1) (udp_connection.cpp:178)
[03:15:53|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:750)
[03:15:53|Debug] Component Autopilot (1) added. (system_impl.cpp:366)
[03:15:53|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[03:15:53|Debug] Discovered 1 component(s) (system_impl.cpp:509)
```
Sometimes it stops there and no crashes happens.
But other times i have

`[03:15:53|Debug] Sending message 76 from 176/224 to 1/1 (mavsdk_impl.cpp:444)`

and it crash immediately.
The backtrace is :

```
#0  0x00007ffff621afe0 in main_arena () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff73ac8bb in std::function<bool (__mavlink_message&)>::operator()(__mavlink_message&) const (this=0x7ffff621afd8 <main_arena+856>, __args#0=...) at /usr/include/c++/11/bits/std_function.h:590
#2  0x00007ffff73a1949 in mavsdk::MavsdkImpl::send_message (this=0x7ffff621ace0 <main_arena+96>, message=...) at MAVSDK/src/mavsdk/core/mavsdk_impl.cpp:453
#3  0x00007ffff7476e0b in mavsdk::ServerComponentImpl::queue_message(std::function<__mavlink_message (MavlinkAddress, unsigned char)>) (this=0x555555a084c0, fun=...) at MAVSDK/src/mavsdk/core/server_component_impl.cpp:170
#4  0x00007ffff7375878 in mavsdk::SystemImpl::queue_message(std::function<__mavlink_message (MavlinkAddress, unsigned char)>) (this=0x7fffbc000e50, fun=...) at MAVSDK/src/mavsdk/core/system_impl.cpp:469
#5  0x00007ffff73c9849 in mavsdk::MavlinkCommandSender::send_mavlink_message (this=0x7fffbc001020, command=std::variant [index 1] = {...}) at MAVSDK/src/mavsdk/core/mavlink_command_sender.cpp:461
#6  0x00007ffff73c8ff2 in mavsdk::MavlinkCommandSender::do_work (this=0x7fffbc001020) at MAVSDK/src/mavsdk/core/mavlink_command_sender.cpp:401
#7  0x00007ffff737483d in mavsdk::SystemImpl::system_thread (this=0x7fffbc000e50) at MAVSDK/src/mavsdk/core/system_impl.cpp:280
#8  0x00007ffff739a7ca in std::__invoke_impl<void, void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> (__f=@0x7fffbc001dd0: (void (mavsdk::SystemImpl::*)(mavsdk::SystemImpl * const)) 0x7ffff7374762 <mavsdk::SystemImpl::system_thread()>, __t=@0x7fffbc001dc8: 0x7fffbc000e50) at /usr/include/c++/11/bits/invoke.h:74
#9  0x00007ffff739a71d in std::__invoke<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> (__fn=@0x7fffbc001dd0: (void (mavsdk::SystemImpl::*)(mavsdk::SystemImpl * const)) 0x7ffff7374762 <mavsdk::SystemImpl::system_thread()>) at /usr/include/c++/11/bits/invoke.h:96
#10 0x00007ffff739a67d in std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> >::_M_invoke<0ul, 1ul> (this=0x7fffbc001dc8) at /usr/include/c++/11/bits/std_thread.h:259
#11 0x00007ffff739a632 in std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> >::operator() (this=0x7fffbc001dc8) at /usr/include/c++/11/bits/std_thread.h:266
#12 0x00007ffff739a612 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> > >::_M_run (this=0x7fffbc001dc0) at /usr/include/c++/11/bits/std_thread.h:211
#13 0x00007ffff64dc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007ffff6094ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#15 0x00007ffff6126850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

Code is basic

```
auto conf = Mavsdk::Configuration{Mavsdk::ComponentType::CompanionComputer};
Mavsdk mavsdk{conf};
auto con_result = mavsdk.add_udp_connection("127.0.0.1", 14550);
if (con_result != ConnectionResult::Success) {
  return false;
}

auto systemOpt = mavsdk.first_autopilot(3.0);
if (!systemOpt.has_value()) {
  return false;
}
```

Not sure what to do at this point.
Thanks for any advices

julianoes, 2024-10-16:
Can you share the example, so I can reproduce it myself?

And maybe you can also share the stacktrace of all threads, in gdb it's `thread apply all bt`.

And you could also run it using `valgrind`, that sometimes tells you more.

depauwjimmy, 2024-10-17:
Of course.
For some reason, doing an example with only code inside main never triggers the message sending. Doing it in a class method does it.
I also have a runloop (ros2 in my case) but it happens with anything that waits so i just put a sleep.

```
#include <mavsdk/mavsdk.h>
#include <unistd.h>

using namespace mavsdk;

class Runtime {
private:
    std::shared_ptr<mavsdk::System> system;

public:
    bool init() {
        auto conf = Mavsdk::Configuration{Mavsdk::ComponentType::CompanionComputer};

        Mavsdk mavsdk{conf};
        auto con_result = mavsdk.add_udp_connection("127.0.0.1", 14550);

        if (con_result != ConnectionResult::Success) {
            return false;
        }

        auto systemOpt = mavsdk.first_autopilot(3.0);
        if (!systemOpt.has_value()) {
            return false;
        }

        this->system = systemOpt.value();
        return true;
    }
};

int main(int argc, char **argv) {
    Runtime runtime;
    runtime.init();
    sleep(10);
    return EXIT_SUCCESS;
}
```

Full stack does not give anything more. There's no other thread in the stacktrace (other than the sleep)

julianoes, 2024-10-21:
Ok, I see the problem. You need to keep `Mavsdk` alive. You can put it in a `std::shared_ptr` or `std::unique_ptr`. In your current implementation it goes out of scope and is destructed.

Also see this comment:
https://github.com/mavlink/MAVSDK/blob/v2.12/src/mavsdk/core/include/mavsdk/mavsdk.h#L35-L36

It might not be the best wording. Probably should be something like:

```
 * An instance of this class must be created and kept alive in order to use the library.
 * The instance can be destroyed after use in order to break connections and release all resources.
```

closed 2024-10-21 by JonasVautherin