mavlink/MAVSDK #2428: MAVSDK mission RAW subscribe_incoming_mission only one mission item input

Wudka-HTWBerlin, 2024-10-21:
Hello,
I’m currently working on a mission sending process for my drones.
I want to send a mission with two or more mission_items.
Unfortunately, I only receive 1 item. I checked the entire sending process and printed the plan that the system will send via GC, and everything seems fine:

On the other side, I checked the receiver, but the mission contains no more than 1 item after being received.
Using the same process on my Pi (as a companion) and sending it to a PX4 works fine.
For telemetry between the GC and the companion, I’m using RFDesign 868x.

Here is the output of the Companion (left) and GC (right).
[image]

Thanks in advance for your help.

Wudka-HTWBerlin, 2024-10-21:
I checked the example of v2.12 and the error also occurs here.

julianoes, 2024-10-21:
Can you give me the steps or example code to reproduce this?

Wudka-HTWBerlin, 2024-10-21:
#include <future>
#include <iostream>
#include <thread>

#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h>
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/param_server/param_server.h>
#include <mavsdk/plugins/param/param.h>
#include <mavsdk/plugins/telemetry_server/telemetry_server.h>
#include <mavsdk/plugins/action_server/action_server.h>
#include <mavsdk/plugins/mission_raw_server/mission_raw_server.h>
#include <mavsdk/plugins/mission/mission.h>

/*
 This example runs a MAVLink "autopilot" utilising the MAVSDK server plugins
 on a seperate thread. This uses two MAVSDK instances, one GCS, one autopilot.

 The main thread acts as a GCS and reads telemetry, parameters, transmits across
 a mission, clears the mission, arms the vehicle and then triggers a vehicle takeoff.

 The autopilot thread handles all the servers and triggers callbacks, publishes telemetry,
 handles and stores parameters, prints received missions and sets the vehicle height to 10m on
 successful takeoff request.
*/

using namespace mavsdk;

using std::chrono::seconds;
using std::chrono::milliseconds;
using std::this_thread::sleep_for;

Mission::MissionItem make_mission_item(
    double latitude_deg,
    double longitude_deg,
    float relative_altitude_m,
    float speed_m_s,
    bool is_fly_through,
    float gimbal_pitch_deg,
    float gimbal_yaw_deg,
    Mission::MissionItem::CameraAction camera_action)
{
    Mission::MissionItem new_item{};
    new_item.latitude_deg = latitude_deg;
    new_item.longitude_deg = longitude_deg;
    new_item.relative_altitude_m = relative_altitude_m;
    new_item.speed_m_s = speed_m_s;
    new_item.is_fly_through = is_fly_through;
    new_item.gimbal_pitch_deg = gimbal_pitch_deg;
    new_item.gimbal_yaw_deg = gimbal_yaw_deg;
    new_item.camera_action = camera_action;
    return new_item;
}

int main(int argc, char** argv)
{
    // We run the server plugins on a seperate thread so we can use the main
    // thread as a ground station.
    std::thread autopilotThread([]() {
        mavsdk::Mavsdk mavsdkTester{
            mavsdk::Mavsdk::Configuration{mavsdk::Mavsdk::ComponentType::Autopilot}};

        auto result = mavsdkTester.add_any_connection("udp://127.0.0.1:14551");
        if (result == mavsdk::ConnectionResult::Success) {
            std::cout << "Connected autopilot server side!" << std::endl;
        }

        auto server_component = mavsdkTester.server_component();

[... 210 lines cut ...]
        std::cout << "Arming failed:" << arm_result << std::endl;
        return 1;
    }

    action.set_takeoff_altitude(10.f);

    // Take off
    std::cout << "Taking off..." << std::endl;
    bool takenOff = false;
    while (true) {
        const Action::Result takeoff_result = action.takeoff();
        if (takeoff_result != Action::Result::Success) {
            std::cout << "Takeoff failed!:" << takeoff_result << std::endl;
            std::this_thread::sleep_for(std::chrono::seconds(1));
            continue;
        }
        break;
    }

    while (true) {
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }

    return 0;
}

julianoes, 2024-10-22:
Thanks, I formatted it with ``` instead of `.

Wudka-HTWBerlin, 2024-11-06:
Hey, just a short reminder on this issue.
Do you have any news on this?

julianoes, 2024-11-06:
Thanks for the reminder.

julianoes, 2024-11-06:
Ok, I tried out the example and in my case I see 7 mission items uploaded, some of them being the waypoints and some the actions like speed change and gimbal change.

So it looks like it's working correctly for me :thinking:...

I tested this on the v2.12 branch, which is v2.12.12.

julianoes, 2025-10-26:
No more answers, closing.

closed 2025-10-26 by julianoes