mavlink/MAVSDK #2434: MAVSDK Debian does not include MAVSDK::mavsdk_action target

Ryanf55, 2024-10-24:
# Description

I want to follow the quick start guide and command an ArduPilot plane to take off, but I cannot compile the recommended example.

## Steps

Follow the quick start here to install the latest release on Ubuntu 22.
https://mavsdk.mavlink.io/main/en/cpp/quickstart.html#install-mavsdk-library

```bash
wget https://github.com/mavlink/MAVSDK/releases/download/v2.12.12/libmavsdk-dev_2.12.12_ubuntu22.04_amd64.deb
sudo dpkg -i ./libmavsdk-dev_2.12.12_ubuntu22.04_amd64.deb
```

Create a `main.cpp` and `CMakeLists.txt` with the following, as per the takeoff guide:
https://mavsdk.mavlink.io/main/en/cpp/guide/taking_off_landing.html#taking-off

```cmake
cmake_minimum_required(VERSION 3.10.2)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
project(mavsdk_takeoff)

find_package(MAVSDK REQUIRED)

add_executable(main main.cpp)
target_link_libraries(main  MAVSDK::mavsdk MAVSDK::mavsdk_action)
```

And, copy the first part of the C++ code
```c++
#include <mavsdk/mavsdk.h>
Mavsdk mavsdk;
ConnectionResult conn_result = mavsdk.add_udp_connection();
// Wait for the system to connect via heartbeat
while (mavsdk.system().size() == 0) {
   sleep_for(seconds(1));
}
// System got discovered.
System system = mavsdk.systems()[0];
```

Build the demo:
```bash
colcon build --packages-up-to mavsdk_demo
```

See the CMake configure error:
```
CMake Error at CMakeLists.txt:9 (add_executable):
  Target "main" links to target "MAVSDK::mavsdk_action" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
  ```

julianoes, 2024-10-24:
Remove `MAVSDK::mavsdk_action`. That's no longer a thing. It's just one `libmavsdk.so` library now, well has been for a while. It's just that the docs are sometimes out of date, sorry about that.

It's actually fixed here:
https://github.com/mavlink/MAVSDK-docs/blob/main/en/cpp/guide/taking_off_landing.md

It's just that the docs generation in CI doesn't seem to run anymore. Sorry about that. I'm hoping to fix that soon.

Ryanf55, 2024-10-24:
Thanks! What about 

```
/home/ryan/Dev/ardupilot_ws/src/mavsdk_demo/main.cpp:9:19: error: ‘class mavsdk::Mavsdk’ has no member named ‘system’; did you mean ‘systems’?
    9 |     while (mavsdk.system().size() == 0) {
 ```

JonasVautherin, 2024-10-24:
> error: ‘class mavsdk::Mavsdk’ has no member named ‘system’; did you mean ‘systems’?

Use `mavsdk.systems()` instead 😉

Ryanf55, 2024-10-24:
Yea, seems like the docs are just out of date. There's missing `using` declarations too.
```
/home/ryan/Dev/ardupilot_ws/src/mavsdk_demo/main.cpp: In function ‘int main()’:
/home/ryan/Dev/ardupilot_ws/src/mavsdk_demo/main.cpp:10:15: error: ‘seconds’ was not declared in this scope; did you mean ‘useconds_t’?
   10 |     sleep_for(seconds(1));
      |               ^~~~~~~
      |               useconds_t
/home/ryan/Dev/ardupilot_ws/src/mavsdk_demo/main.cpp:10:5: error: ‘sleep_for’ was not declared in this scope
   10 |     sleep_for(seconds(1));
```

Perhaps the tutorials could be updated such that you can copy-paste the code without changes and have it compile?

julianoes, 2024-10-25:
Agreed. Feel free to make pull requests. I'm meaning to do a pass through the docs but it will be after the v3 release.

closed 2025-06-02 by julianoes