mavlink/MAVSDK #2436: When sending motion capture data to two drones using set-visionposition estimate, both drones obtained the same position data regardless of whether their sysids were the same.

yuxuanwu1234567, 2024-11-05:
The local data of the first and second aircraft are completely chaotic.
[image]
[image]
[image]
[image]

yuxuanwu1234567, 2024-11-06:
No matter how many planes I connect, system 245 appears.

yuxuanwu1234567, 2024-11-06:
[image]

julianoes, 2024-11-06:
Do you mind sharing (some of) your code so I can understand what you're doing?

julianoes, 2025-06-02:
OP never came back with steps to reproduce this, closing.

closed 2025-06-02 by julianoes