mavlink/MAVSDK #2437: Emergency! help help！！！！I am from China, you are a great and lovely group of people, I have a problem now,

liuchuanping, 2024-11-08:
[image]
[image]

liuchuanping, 2024-11-08:
We want to open a ground station platform using java/vue.js. This platform can manage multiple drones. And the platform can control different drones and obtain data from different drones. This flowchart is a traditional solution, but I think the development workload is too large (netty+mavlink+tcp/mqtt). I want to use your mavsdk_server. I don’t know much about c++. Now I only compile and run mavsdk_server, but I found many problems when I tested it through java api (io.mavsdk)?
1. User A sends a takeoff command to drone 1, so how to ensure that only drone 1 can receive it. Has mavsdk done all these solutions?
2. As you said, if mavsdk_server can solve the problem of controlling multiple drone connections by running multiple instances, then how to solve the port problem? Because the ground station backend can only provide fixed ports to drones (to be precise, it is an embedded box mounted on the drone. This box is a separately developed circuit board, and the configuration of each box is fixed)
3. And most importantly, I want to hear your ideas! I don't know if mavsdk is still suitable for me with such requirements? Thank you very much!
4. Another point is what mavsdk does compared with traditional drone communication development (netty+mavlink+tcp/mqtt).

julianoes, 2024-11-10:
Hi!

1./2. mavsdk_server only supports one drone. This means you need to start multiple mavsdk_server instances, one per drone.
3. I hope so.
4. What's netty? MAVSDK basically makes some of the protocols on top of MAVLink easier. E.g. it handles the mission upload or FTP transfer.

liuchuanping, 2024-11-11:
> Hi!
> 
> 1./2. mavsdk_server only supports one drone. This means you need to start multiple mavsdk_server instances, one per drone. 3. I hope so. 4. What's netty? MAVSDK basically makes some of the protocols on top of MAVLink easier. E.g. it handles the mission upload or FTP transfer.

very thanks! Friends from a distant land
1.  so i think i can run mavsdk_server with a docker container , then every docker container operates a drone, 
2. i think that is a good idea! just one question: The port could be dynamic. But for drones, it doesn't want that to happen. 
3.. netty is High performance and high concurrency communication framework for java!

liuchuanping, 2024-11-11:
I really long for your free country.

julianoes, 2024-11-11:
> The port could be dynamic.

Which port would be dynamic?

Maybe you could you change the port that PX4 sends MAVLink to, and then route it to the right docker container.

liuchuanping, 2024-11-11:
> > The port could be dynamic.
> 
> Which port would be dynamic?
> 
> Maybe you could you change the port that PX4 sends MAVLink to, and then route it to the right docker container.

PX4 will not be used. The production environment is directly connected to the drone (to be precise, the ARM device mounted under the drone cabin).
My idea is that the ARM device on the drone will first log in and connect to my ground station web platform (a fixed nginx proxy port) by sending an http request. Then http will return the port of mavsdk_server (dynamic ports will not be affected. udp or tcp port, for example: 14540), and then the arm device will establish a connection based on this mavsdk_server port (tcp or udp, for example: 14540). Send or receive mavlink data packets
Of course, my remaining questions are:
1. Does mavsdk_server support both tcp/udp for port 14540?
2. The difficulty is that the ARM device mounted on the drone needs to communicate with mavsdk_server (14540). This is a bit troublesome for ARM device developers. It cannot be directly connected to the background Java program, but indirectly through mavsdk_server (14540). For me, mavsdk_server is a heiha?

julianoes, 2024-11-11:
MAVSDK does support TCP as well, however, TCP server conditions are only in the main branch, and not in v2.12 (see https://github.com/mavlink/MAVSDK/pull/2342).

liuchuanping, 2024-11-12:
> MAVSDK does support TCP as well, however, TCP server conditions are only in the main branch, and not in v2.12 (see #2342).

thanks my bro

closed 2025-05-20 by julianoes