mavlink/MAVSDK #2445: Detecting button pressing in real radio command with MAVSDK

jonazti07, 2024-11-20:
I'm a new user of MAVLINK and I'm looking for a particular functionality that I'm not sure if it's implemented in MAVSDK (i.e, I haven't seen it in the source code).

I'm trying to detect the actual pressing of a button in the real radio command of my drone.

Is this possible with MAVSDK?

julianoes, 2024-11-20:
Ok, can you give some more context? So, where is MAVSDK used: on the ground or on a companion? In other words, do you want to detect the button pressed on the drone based on what a ground station (or RC) sends? Or do you want a button command to be sent from the ground using MAVSDK?

jonazti07, 2024-11-21:
MAVSDK is used in a companion computer (Jetson). And yes, I want to detect the button pressed on the drone based on what a ground station (or RC) sends.
Let me know if you need further explanations.

julianoes, 2024-11-21:
Ok, and what message is used to send that button? Presumably it's [MANUAL_CONTROL](https://mavlink.io/en/messages/common.html#MANUAL_CONTROL)?

In that case, I wonder if PX4 emits something to signal that a button has been pressed, presumably not, and you have to "intercept" this message in MAVSDK.

Are you using MAVSDK in C++? In that case you can intercept messages manually:
https://github.com/mavlink/MAVSDK/blob/v2.12/src/mavsdk/core/include/mavsdk/mavsdk.h#L408-L420

jonazti07, 2024-11-22:
I think there has been a misunderstanding. I'm not sending myself the buttons states, but I want to receive them.
Yes, I'm ussing MAVSDK in C++, but I think this funcitonality is not currently supported, so I'll need to detect this flag in another way in my program.
Thanks!

julianoes, 2024-11-25:
So what you're saying is that you're not sending MAVLink RC to PX4, so there is no way for MAVSDK to intercept that message. In that case you have to make sure PX4 sends out [RC_CHANNELS](https://mavlink.io/en/messages/common.html#RC_CHANNELS) and than subscribe to this one in MavlinkPassthrough.

jonazti07, 2024-11-25:
Okay, I see. Thanks for your support.

julianoes, 2024-11-25:
You're welcome. I'll close this then.