mavlink/MAVSDK #2448: log_capture feature has some problem, my guess.

wuyong-ndrone, 2024-11-21:
### Environment

- WSL debian bookwarm
- MAVSDK 2.12.12

### Issues

I want to use the "log_capture" feature, and found some problems. My log is created between arm and disarm.

#### first log is good but after, not good

After the my program is started, the first log file is good. But after the second log file shows, (PlotJuggler)
[image]
And in HEX editor shows like this,
[image]
uLog file's header is gone. Again, first ulog file is ok.

So, I tried different way like this.

```cpp
    if (_system)
    {
        if (enable) // start the log capture
        {
            _log_streaming       = std::make_shared<mavsdk::LogStreaming>(_system);
            _h_log_streaming_raw = _log_streaming->subscribe_log_streaming_raw(
                std::bind(&AutoPilot_Impl::onLogStreamingRaw, this, std::placeholders::_1));
            _log_streaming->start_log_streaming();
        }
        else // stop
        {
            _log_streaming->stop_log_streaming();
            _log_streaming->unsubscribe_log_streaming_raw(_h_log_streaming_raw);
            _log_streaming.reset();
        }
    }
```

In this case, second log file was good. ulog header is placed on start of file. But try to start the third log file, program is stuck like this.
[image]

#### Am I did something wrong?

wuyong-ndrone, 2024-11-21:
Hi,
I found some problems in second log file.

[image]

After the second log file, some garbage is added in top of new file's header.

julianoes, 2024-11-21:
I'm using log streaming with this version and it works for me, so my hunch is that you write invalid data somewhere. Can you share more of the code?

wuyong-ndrone, 2024-11-22:
When mavlink is connected

```cpp
    _h_log_streaming_raw = _log_streaming->subscribe_log_streaming_raw(
        std::bind(&AutoPilot_Impl::onLogStreamingRaw, this, std::placeholders::_1));
```

And, callback is

```cpp
void AutoPilot_Impl::onLogStreamingRaw(mavsdk::LogStreaming::LogStreamingRaw log_raw)
{
    if (sigLogStreamingUpdated.num_slots() > 0)
    {
        auto log = mavsdk::base64_decode(log_raw.data);
        sigLogStreamingUpdated(log);
        // sigLogStreamingUpdated(log_raw.data);
    }
}
```

Log streaming start/stop

```cpp
void AutoPilot_Impl::SetLogCapture(bool enable)
{
    if (_log_streaming)
    {
        if (enable)
            _log_streaming->start_log_streaming();
        else
            _log_streaming->stop_log_streaming();
    }
}
```

Before the log capture module is started,

```cpp
    // register callback
    _config.autopilot_->SigLogStreamingUpdatedHandle().connect(
        boost::bind(&LogCapture_Impl::slotAutoPilotLogStreamingUpdated, this, boost::placeholders::_1));
```

Callback handler and circular buffer

```cpp
// in header file
    boost::circular_buffer<std::vector<uint8_t>> _ring_buffer { 10 };

// and callback handler
void LogCapture_Impl::slotAutoPilotLogStreamingUpdated(std::vector<uint8_t>& log)
{
    std::lock_guard<std::mutex> locker(_ring_buffer_lock);
    _ring_buffer.push_back(log);
}
```

Here is log to file

```cpp
void LogCapture_Impl::run()
{
    bool          logging { false };
    std::ofstream log_file;
    std::vector<uint8_t> log_row;

    // for stabilization delay
    std::this_thread::sleep_for(std::chrono::milliseconds(500));

    try
    {
        if (not std::filesystem::is_directory(_config.file_path_))
            std::filesystem::create_directories(_config.file_path_);
    }
    catch (std::exception& e)
    {
#ifndef NDEBUG
        std::cerr << e.what() << std::endl;
#endif
        _config.file_path_ = DEFAULT_LOGDIR;
        std::filesystem::create_directories(std::filesystem::path(_config.file_path_));
    }

    while (not _run_thread->interruption_requested())
    {
        if (logging != _start_logging)
        {
            logging = _start_logging;

            if (logging)
            {
                std::string file_path { _config.file_path_ + "/" + createFileName() };
                log_file.open(file_path, std::ios::binary);
                _config.autopilot_->SetLogCapture(true);
            }
            else
            {
                if (log_file.is_open())
                {
                    _config.autopilot_->SetLogCapture(false);
                    log_file.close();
                }
                _ring_buffer.clear();
            }
        }

        if (log_file.is_open() and (not _ring_buffer.empty()))
        {
            {
                std::lock_guard<std::mutex> locker(_ring_buffer_lock);
                log_row = _ring_buffer.front();
                _ring_buffer.pop_front();
            }

            log_file.write(reinterpret_cast<char*>(log_row.data()), log_row.size());
        }
        else
        {
            std::this_thread::sleep_for(std::chrono::milliseconds(2));
        }
    }
}
```

julianoes, 2024-11-25:
Thanks. Looks correct from what I can see. The version I used/tested recently was the one on v2.12. So maybe I broke something when going to the base64 implementation. I will have to test this one when I upgrade to future v3.

wuyong-ndrone, 2024-11-25:
I think the log is no need the base64 encoding. I think it just increase data size. If need the base64 encoding, it should process the user. Just my opinion.
Thank you for your fast reply.

julianoes, 2024-11-25:
The problem is that we need to do base64 encoding for language wrappers like Python, because we can't send binary data there using the string type.

julianoes, 2025-06-02:
This is switched to base64 with v3, closing