mavlink/MAVSDK #2450: Timed out waiting for system

Steven-Ajex, 2024-11-25:
I'm using MAVSDK to run some test,here is how i do it:
1.I connected the PC with my flight controller through serial
2.I run : `build/takeoff_and_land serial:///dev/ttyUSB0:9216`

most of the time it will print
`[05:05:55|Info ] MAVSDK version: v2.12.2-136-ga81d2588-dirty (mavsdk_impl.cpp:28)
Timed out waiting for system
`
but it will always work if i rerun it again
`[05:05:55|Info ] MAVSDK version: v2.12.2-136-ga81d2588-dirty (mavsdk_impl.cpp:28)
[05:05:55|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:743)
[05:05:55|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[05:05:55|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[05:05:55|Debug] Discovered 1 component(s) (system_impl.cpp:497)
[05:05:55|Debug] Request message 148 using REQUEST_MESSAGE (mavlink_request_message.cpp:78)`

"Does anyone know how to solve this?Thanks a lot"

julianoes, 2024-11-25:
`9216` doesn't look like a real baudrate?

julianoes, 2025-03-17:
No more follow up, closing.

closed 2025-03-17 by julianoes