mavlink/MAVSDK #2452: How to use the `set_always_send_heartbeats`` of Configuration

flaviol-souza, 2024-11-26:
I'm building on a sample Proxy Mavlink with MAVSDK between QGC and PX4 (SITL).
I would like to know how to use the ``set_always_send_heartbeats`` of Configuration for Autopilot even if it is defined as false. 
`Mavsdk::Configuration custom_config(1, 1, false);
 custom_config.set_always_send_heartbeats(false);
 Mavsdk mavsdk_px4{custom_config}; `

This attribute is reset to true in MavsdkImpl::add_udp_connection when the ``add_any_connection`` is called (which seems to me to be a bug).

[image]

julianoes, 2024-11-27:
The configuration option is set internally depending on the connection you add. If you are sending UDP messages to another IP (instead of listening on a local port) you need to send something to initiate the connection and allow the counterpart to send anything back. Otherwise, the communication can't really start.

What are you trying to do exactly? What sort of connections are you adding?

flaviol-souza, 2024-11-27:
@julianoes 
I am creating a "MAVLink Proxy" using MAVSDK. The primary purpose of the Proxy is to act as a MAVLIN message broker between QGroundControl (QGC) and PX4 SITL. This proxy intercepts all MAVLink messages sent between the two components (QGC and PX4) via ``subscribe_message`` of MavlinkPassthrough. Later, when the other component connects, the message is forwarded to the destination.

The basic flow is:

1. QGC connects to the Proxy, but believes it is connected directly to PX4 (`Mavsdk::Configuration mavsdk_px4{Mavsdk::Configuration{Mavsdk::ComponentType::Autopilot}};`).
2. The PX4 connects to the Proxy, but believes it is connected directly to the QGC (`Mavsdk mavsdk_qgc{Mavsdk::Configuration{Mavsdk::ComponentType::GroundStation}};`).
3. The Proxy transparently forwards MAVLink messages between the two.

**Challenges Found**

1. **Automatic HEARTBEAT Messages from the Proxy (MAVSDK)**:
- When I configure MAVSDK for `ComponentType::Autopilot`, it sends automatic `HEARTBEAT` messages to initiate communication with the QGC.
- These `HEARTBEAT` messages generated by the Proxy (with default values, such as `type == 0`) are mixed with the real messages from the PX4, causing problems in the QGC behavior. For example, the QGC constantly switches between the "PreFlight" and "Custom" states. That's why the QGC doesn't understand that the PX4 is ready for flight, releasing the Take Off commands, for example.

2. **Internal Behavior of `set_always_send_heartbeats`**:
- I explicitly configured `set_always_send_heartbeats(false)` in `Mavsdk::Configuration`. However, when the `add_any_connection` function is called, the attribute is reset to `true` in the `add_udp_connection` method due to the internal logic of MAVSDK.
I understand the justification for sending `HEARTBEAT` when adding an outgoing UDP connection, but in this case, the Proxy should act as a passive (reactive) intermediary. In other words, it should not send `HEARTBEAT` messages automatically; it should just redirect the real messages from the PX4 to the QGC and vice versa.

My goal is to completely disable the sending of `HEARTBEAT` messages automatically generated by MAVSDK in the Proxy. I would like:
- The Proxy is to send MAVLink messages to the QGC only when it receives a message from the PX4, and vice versa.
- No `HEARTBEAT` messages are generated or sent by the Proxy in normal situations.

julianoes, 2024-11-28:
I see, thanks for the context. That's quite involved.

I can see how this automatic option does not work for you, but I also don't want to break the behavior for existing users.

One workaround I can think of is to intercept the outgoing messages and drop the heartbeats manually, using:
https://github.com/mavlink/MAVSDK/blob/v2.12/src/mavsdk/core/include/mavsdk/mavsdk.h#L422-L434

Would that work?

flaviol-souza, 2024-12-11:
Thank you for the suggestion! 

I tested the approach of intercepting outgoing messages and manually dropping the heartbeats using the `intercept_outgoing_messages_async()` method, and it worked perfectly for my use case. I implemented an MAVLink proxy that forwards messages between PX4 and QGroundControl while filtering specific messages as needed.

To help others with similar requirements, I have created an example and submitted it as a pull request: [#2467](https://github.com/mavlink/MAVSDK/pull/2467). This example demonstrates how to set up an MAVLink proxy using `MavlinkPassthrough` and includes message interception and forwarding logic.

Thank you again for pointing me in the right direction!

closed 2024-12-11 by flaviol-souza