mavlink/MAVSDK #2456: Having issues with installing mavsdk.

Kareem3807, 2024-12-02:
Hey, I'm on Mac,

I used the install instructions on the website , using homebrew.

I installed mavsdk via home-brew and that went sucessfully, however when trying to compile anything using Mavsdk (I'm using the battery example). I get this error:

Mavlink.cpp:6:10: fatal error: 'mavsdk/mavsdk.h' file not found
#include <mavsdk/mavsdk.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.

julianoes, 2024-12-02:
What's your cmake file to get mavsdk?

Kareem3807, 2024-12-02:
How can I check? As far as I know I haven't done anything with Cmake, my apologies, I'm new to this. what should I do?

julianoes, 2024-12-02:
How are you building your C++ file?

Kareem3807, 2024-12-02:
I'm using Vscode and just compiling the file like so:

g++ -arch arm64 -std=c++17 Mavlink.cpp

julianoes, 2024-12-03:
Ok, when you compile C++, you need to learn how to set include directories (so it finds the includes) and link to libraries. That's a general C++ concept, and nothing MAVSDK specific.

MAVSDK tries to be like any other library, and supports specifically cmake (and its find_package).

Best to just look at how examples are built, e.g.:
https://github.com/mavlink/MAVSDK/blob/main/examples/gimbal/CMakeLists.txt

There is also some documentation for this:
https://mavsdk.mavlink.io/main/en/cpp/guide/toolchain.html#build-definition-file---cmakeliststxt

Unfortunately, it's out of date, so linking to the plugins is no longer needed, so:

```
target_link_libraries(your_executable_name
    MAVSDK::mavsdk             # needed!
    MAVSDK::mavsdk_action      # not needed, remove this!
)

```

closed 2024-12-03 by julianoes