mavlink/MAVSDK #2474: Missing Chunks and Payload Offset Errors During Download. Cannot download any files completely using MAVSDK Java in android studio.

AtiqAakash, 2024-12-17:
### Title: **Missing Chunks and Payload Offset Errors During Download**

#### Description:
I'm trying to make an android app to download the log files from the cubepilot. So I tried the mavsdk examples for downloading log file. I also tried mavsdk ftp to download any files located in the cube. But cannot download any file completely. 

I'm experiencing an issue with payload offsets and missing chunks during the download process in our app using MAVSDK. Specifically, we are seeing discrepancies in the payload offsets, with the warning message:

io.mavsdk.androidclient              W  Got payload offset: 4541, next offset: 5736

Additionally, we are getting missing chunk errors that seem to occur during the download of data.

#### Expected Behavior:
The download process should handle the payloads correctly, with consistent offsets and no missing chunks during the data transfer.

#### Actual Behavior:
The payload offsets are inconsistent, and there are missing chunks, resulting in errors and incomplete downloads. 

I'm trying the app in herelink controller.

Any help or insight into resolving this issue would be greatly appreciated!

julianoes, 2024-12-18:
Thanks for the issue.

> I'm trying to make an android app to download the log files from the cubepilot.

I will have to try to reproduce this, so I need to replicate your setup. Can you tell me what software you run on the CubePilot and what version and how you connect to it?

AtiqAakash, 2024-12-18:
Thank you for your reply and for looking into this issue. Here are the details of my setup:

CubePilot Software and Version:

I am using the CubePilot Orange/Blue (mention your specific model).
Firmware: PX4 versions tested include:
1.14.0
1.15.0
1.15.2
Connection Details:

I am connecting to the CubePilot using a Herelink controller.
The Herelink runs Android 7.1.2.
The connection is established via telemetry radio or WiFi (mention your exact connection method).
Android App Details:

App developed with MAVSDK version 2.1.0 and MAVSDK server version 2.1.13.
Minimum SDK version: 21.
Using Java for development.
I can also paste my code here if that would help you debug further. Please note that I’m completely new to Android development and Java, so it’s possible I might have implemented something incorrectly. Your guidance would be greatly appreciated!

julianoes, 2024-12-18:
Thanks for the info! I suspect this is within the C++ library, so I'll try to reproduce without Herelink first and then add more things in if required.

julianoes, 2024-12-19:
Trying to reproduce this, so downloading some ulog files with PX4 v1.15.3+, on `main` branch:

- Pixhawk connected via USB: :heavy_check_mark:
- Pixhawk Telem1 connected via FTDI cable at 57600 baudrate: :heavy_check_mark: 
- Pixhawk Telem1 via SiK radio: :heavy_check_mark:
- Pixhawk Telem1 via Herelink -> Tethering back to desktop: :heavy_check_mark: 

Now, on `v2` branch:

`- Pixhawk Telem1 via Herelink -> Tethering back to desktop: :heavy_check_mark: `

I'm wondering what version you're exactly using:

> MAVSDK server version 2.1.13

This version doesn't exist. Given you're MAVSDK-Java version is 2.1.0 would mean that you're likely on server [v2.12.2](https://github.com/mavlink/MAVSDK-Java/blob/aee2c1e3f2e3fb7a4625e51be04065d1fe931a55/mavsdk_server/build.gradle#L22)

The last fix I can see would be:
https://github.com/mavlink/MAVSDK/pull/2235
which went into v2.5.0, so you should have that one.

I'm not sure what exactly is going on to be honest :confused:.

I would suggest that you try with the latest version of MAVSDK-Java which updates mavsdk_server under the hood. If it still happens there, it would be great if you could create a small example (can be a fork of the MAVSDK-Java example) for me to run and debug on Herelink. Thanks.

AtiqAakash, 2024-12-19:
Hello Julianoes,

Thank you so much for your super fast effort!

I'll definitely try that and update you. Unfortunately this is my last day at office before the vacation and all my development environment is in the office. So unfortunately I have to try this and update you next year. 

Til then happy christmas!

julianoes, 2024-12-20:
No worries, happy holidays!

julianoes, 2025-06-02:
Assuming this was fixed, closing.

closed 2025-06-02 by julianoes