mavlink/MAVSDK #2475: pkg-config file has CI paths

battlesnake, 2024-12-17:
```
mark@soovana:~/repos/sim_tracker $ pkg-config --cflags mavsdk
-I/home/runner/work/MAVSDK/MAVSDK/install/include -I/home/runner/work/MAVSDK/MAVSDK/install/include/mavsdk -I/usr/include/x86_64-linux-gnu -I/usr/include/jsoncpp
```

The include paths in the current ubuntu 22.04 pkg-config file seem to reference CI user directories instead of the eventual installation directories.

battlesnake, 2024-12-17:
```
$ cat /usr/lib/pkgconfig/mavsdk.pc
prefix=/home/runner/work/MAVSDK/MAVSDK/install
exec_prefix=/home/runner/work/MAVSDK/MAVSDK/install/bin
libdir=/home/runner/work/MAVSDK/MAVSDK/install/lib
includedir=/home/runner/work/MAVSDK/MAVSDK/install/include
```

battlesnake, 2024-12-17:
Similar for Debain 12 ARM64: `/work/build/linux-arm64/install`

JonasVautherin, 2024-12-17:
[This](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk.pc.in) is what's being used to generate it. If I remember correctly, those variables are defined with `CMAKE_INSTALL_PREFIX` and [GNUInstallDirs](https://cmake.org/cmake/help/v3.16/module/GNUInstallDirs.html).

My guess would be that it is all coming from `CMAKE_INSTALL_PREFIX` in the end. And it is set e.g. [here](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L249) in the CI.

I feel like the CI jobs for the CI should not override `CMAKE_INSTALL_PREFIX` and install on the system instead.

Would you be able to test that?

battlesnake, 2024-12-17:
Sure I can test stuff - I'm installing MAVSDK as part of a container build so I can test changes pretty easily by applying them and commenting out my current workarounds.  I'll try to fit it in sometime tomorrow.

julianoes, 2024-12-18:
Those paths work for the .deb packages, so I'm confused why they don't for the pkg-config. I guess we always just use CMake's `find_package` and not pkg-config.

closed 2025-01-26 by julianoes