mavlink/MAVSDK #2483: how can i resolve this problem? please

liuchuanping, 2024-12-20:
My flight control module is connected to mavsdk_server through the serial port. I start mavsdk_server ( ./mavsdk_server -p 50051 serial:///dev/ttyUSB0:115200 ), but an error is reported. The exception log is as follows:  
 
[10:42:54|Info ] MAVSDK version: v2.12.2-126-gcebb708a (mavsdk_impl.cpp:28)
[10:42:54|Info ] Waiting to discover system on serial:///dev/ttyUSB0:115200... (connection_initiator.h:20)
[10:42:59|Debug] New system ID: 50 Comp ID: 1 (mavsdk_impl.cpp:736)
[10:42:59|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[10:42:59|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[10:42:59|Debug] Discovered 1 component(s) (system_impl.cpp:497)
[10:42:59|Debug] Request message 148 using REQUEST_MESSAGE (mavlink_request_message.cpp:78)
[10:42:59|Info ] System discovered (connection_initiator.h:62)
[10:42:59|Info ] Server started (grpc_server.cpp:173)
[10:42:59|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:174)
[10:43:00|Warn ] sending again after 0.500474 s, retries to do: 3  (512). (mavlink_command_sender.cpp:322)
[10:43:00|Warn ] Request was for msg ID: 148 (mavlink_command_sender.cpp:328)
[10:43:01|Warn ] sending again after 1.00707 s, retries to do: 2  (512). (mavlink_command_sender.cpp:322)
[10:43:01|Warn ] Request was for msg ID: 148 (mavlink_command_sender.cpp:328)
[10:43:01|Warn ] sending again after 1.51445 s, retries to do: 1  (512). (mavlink_command_sender.cpp:322)
[10:43:01|Warn ] Request was for msg ID: 148 (mavlink_command_sender.cpp:328)
[10:43:02|Error] Retrying failed for REQUEST_MESSAGE command for message: 148, to (50/1) (mavlink_command_sender.cpp:361)

julianoes, 2024-12-21:
That's likely just warnings you can ignore. Are you using PX4 or ArduPilot? And are you trying anything that's not working?

liuchuanping, 2024-12-24:
Error] Retrying failed for REQUEST_MESSAGE command for message: 148, to (50/1) (mavlink_command_sender.cpp:361)
is this a warning?

liuchuanping, 2024-12-24:
I used MAVSDK to communicate with a flying module device mounted on a drone. This module device is a product I bought from a drone flight control hardware manufacturer. The manufacturer said that the module supports the MAVLink protocol, and I successfully simulated it using their own ground station and non air and simulation equipment. However, when I connected their flight control to MAVSDK, there was a problem. The exception message is as follows: Error] Retrying failed for REQUEST_MESSAGE command for message: 148, to (50/1) (mavlink_command_sender.cpp:361)

julianoes, 2024-12-27:
Message 148 is [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION).

It sounds like this module is implementing MAVLink but might not be implementing the REQUEST_MESSAGE command yet.

Is this module "faking" to be a drone?

On the MAVSDK side, we probably have to fall back to the old command which is [MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES). That might help. I can try that some time.

Does it work at all, or not? I suspect not because it doesn't see an autopilot unless it gets that message.

julianoes, 2025-01-25:
No more answers, closing.

closed 2025-01-25 by julianoes