mavlink/MAVSDK #2486: Disconnection Issue with Ardupilot Flight Controller on serial port

yangqunfeng, 2024-12-28:
The C++ mavsdk-based onboard computer program connects to the Ardupilot flight controller through Telem2. After establishing the connection, the flight controller will automatically disconnect after a random duration. Later, I discovered that the flight controller’s Telem2 serial port had no data output. However, after restarting the flight controller software using Mission Planner (MP), data returned and the connection was re-established correctly (I ruled out hardware issues through this). Once the serial port detects that the flight controller has disconnected, even if I restart my control program, it still cannot reconnect. I later tried using a TCP/UDP connection to the flight controller, but the disconnection issue persisted. However, sometimes after about half an hour, the flight controller reconnects on its own. This issue has been bothering me for months. I want to know why the flight controller serial port behaves this way. From the Ardupilot wiki, I learned that there may be conflicts caused by telemetry link sharing, but I am not sure if this is the cause of the problem, since I also use Telem1 to connect to QGC. I would greatly appreciate it if anyone could help me analyze the possible causes.

julianoes, 2024-12-29:
I have not heard of this problem before.

Can you tell me all the versions involved in this? ArduPilot and MAVSDK, and what hardware. And any other relevant settings. Or if you have exact steps for me to reproduce the problem, even better.

yangqunfeng, 2024-12-29:
**Ardupilot** is using Copter-4.5.6, and below are the specific parameters of the flight controller.
[6011.zip](https://github.com/user-attachments/files/18268933/6011.zip)

**MAVSDK** is currently using versions 2.12.3 and 2.13.0.
The **hardware** is CUAV X7+, with a custom expansion board.
The logic of my code is as follows:
First, I use the **subscribe_on_new_system** function, which is used to callback when the flight controller connects and disconnects. Then, I create a telemetry service, but also connect to the drone through passthrough, and I set the message intervals for the following messages using MAV_CMD_SET_MESSAGE_INTERVAL. The key is the message ID, and the value is the message interval.
`const std::vector<std::pair<uint32_t, float>> message_intervals = {
        {MAVLINK_MSG_ID_GPS_RAW_INT, 500000.0f},
        {MAVLINK_MSG_ID_GLOBAL_POSITION_INT, 500000.0f},
        {MAVLINK_MSG_ID_ATTITUDE, 500000.0f},
        {MAVLINK_MSG_ID_MISSION_CURRENT, 1000000.0f},
        {MAVLINK_MSG_ID_BATTERY_STATUS, 1000000.0f},
        {MAVLINK_MSG_ID_STATUSTEXT, 1000000.0f},
        {MAVLINK_MSG_ID_POWER_STATUS, 1000000.0f},
};
`
I have created the corresponding callback functions to receive messages such as **Global Position** and **Raw Position**, and I check whether the position information is being returned at the specified frequency. By the way (a side note), when I previously used MAVSDK version 2.0.1, when I detected that the drone hadn't sent the subscribed messages for a while, I would again use MAV_CMD_SET_MESSAGE_INTERVAL to resubscribe to the messages. Although the drone would occasionally disconnect, it didn't happen as frequently as it does now.
I am sure the issue might be with my program, mainly because the QGC (Ground Control Station) doesn't show the drone disconnecting, but my program does.
I also want to ask if the message intervals I’ve set are feasible, particularly {MAVLINK_MSG_ID_GPS_RAW_INT, 500000.0f} and {MAVLINK_MSG_ID_GLOBAL_POSITION_INT, 500000.0f}. I previously encountered situations where gps_fix_type was no_gps and global_position_health was false, but it was actually rtk_fix. I improved this issue by increasing the message interval and having MP (Mission Planner) actively send message groups.
Thank you for your response as the main maintainer of the MAVSDK project. This is very important to me. I appreciate your patient explanation.

yangqunfeng, 2024-12-30:
Today, I discovered again that the serial port automatically reconnects after about 10 minutes, with the following prompt:
Discovered 1 component(s) (system_impl.cpp:509)  
Discovered 2 component(s) (system_impl.cpp:509)  
User callback queue too slow. See: https://mavsdk.mavlink.io/main/en/cpp/troubleshooting.html#user-callbacks (mavsdk_impl.cpp:832)

julianoes, 2025-01-05:
Could you create a small example to reproduce this?

yangqunfeng, 2025-01-09:
The issue was eventually traced to a potential problem with the custom flight controller's onboard circuit board. I ran the program for an extended period on a simulator without any issues, so this was a false alarm. Apologies for wasting everyone's time.

closed 2025-01-09 by yangqunfeng