mavlink/MAVSDK #2488: SSL error building library on Windows in configure step

hamishwillee, 2025-01-03:
Clean build in windows command prompt after following these instructions https://mavsdk.mavlink.io/main/en/cpp/guide/build.html#windows to install VS tools and cmake.

OpenSSL also seems to cause problems building on Ubuntu.

Configure step finished with

```

  No update step for 'openssl'
  No patch step for 'openssl'
  Performing configure step for 'openssl'
  'perl' is not recognized as an internal or external command,
  operable program or batch file.
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5): error MSB8066: Custom build for 'D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-mkdir.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-download.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-update.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-patch.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-configure.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-build.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\4cc26649d8cfd21648a879eb2f147cb5\openssl-install.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\b8e56295dec8820efe4bf03ee24ddcb8\openssl-complete.rule;D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\CMakeFiles\642fad925227dfd8d73fced3e3738b0d\openssl.rule;D:\github\mavsdk\MAVSDK\third_party\openssl\CMakeLists.txt' exited with code 9009. [D:\github\mavsdk\MAVSDK\build\default\third_party\openssl\openssl.vcxproj]
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  D:/github/mavsdk/MAVSDK/build/default/third_party/openssl failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:20 (build_target)

-- Configuring incomplete, errors occurred!
````

Full build log at https://gist.github.com/hamishwillee/a8e54796c88a2a18c1f0b3d738a291b0

JonasVautherin, 2025-01-03:
It fails [here](https://gist.github.com/hamishwillee/a8e54796c88a2a18c1f0b3d738a291b0#file-windows-mavsdk-build-fail-L1247):

> 'perl' is not recognized as an internal or external command, operable program or batch file.

Did it use to build, or is it the first time you try on this computer? I don't think we have changed anything for Windows and OpenSSL, so I wonder if we don't "just" require perl to be installed (as a "dev tool")? Not sure :thinking:

JonasVautherin, 2025-01-03:
It seems to me that Windows needs Perl for OpenSSL: https://github.com/openssl/openssl/blob/master/NOTES-PERL.md#perl-on-windows.

I found some unofficial instructions [here](https://github.com/kiyolee/openssl3-win-build/blob/main/NOTES-WINDOWS.md#native-builds-using-visual-c).

julianoes, 2025-01-26:
I just tried this on my Win 10 installation with latest cmake 3.431.4, Visual Studio 17 2022, and perl 5 installed.

JonasVautherin, 2025-01-26:
Did it work, or did you reproduce the issue?

julianoes, 2025-01-26:
Blame my fever. It compiled fine.

julianoes, 2025-01-26:
Leaving it open until it works for @hamishwillee as well.

julianoes, 2025-03-17:
Closing, re-open @hamishwillee if it's still not working.

closed 2025-03-17 by julianoes