mavlink/MAVSDK #2489: SSL error building/configuring library on Ubuntu 22.04 in WSL2 and building docs

hamishwillee, 2025-01-03:
The build log for the docs is linked below. This is latest main as per today, set up as per the build instructions on Ubuntu 22.04. The environment was not completely clean, but the build was. Same error if I do the configure:

https://gist.github.com/hamishwillee/4b23a2ca97fb587a6e356bddb5ec63dd

Last few lines are:

```
[ 99%] Building CXX object src/integration_tests/CMakeFiles/integration_tests_runner.dir/mavlink_passthrough.cpp.o
[100%] Building CXX object src/integration_tests/CMakeFiles/integration_tests_runner.dir/mission_transfer_lossy.cpp.o
[100%] Linking CXX executable integration_tests_runner
/usr/bin/ld: ../mavsdk/libmavsdk.so.2.12.2: undefined reference to `SSL_get0_group_name'
collect2: error: ld returned 1 exit status
make[2]: *** [src/integration_tests/CMakeFiles/integration_tests_runner.dir/build.make:694: src/integration_tests/integration_tests_runner] Error 1
make[2]: Leaving directory '/home/ubuntu/github/mavsdk/MAVSDK/tools/docs'
make[1]: *** [CMakeFiles/Makefile2:918: src/integration_tests/CMakeFiles/integration_tests_runner.dir/all] Error 2
make[1]: Leaving directory '/home/ubuntu/github/mavsdk/MAVSDK/tools/docs'
make: *** [Makefile:136: all] Error 2
make: Leaving directory '/home/ubuntu/github/mavsdk/MAVSDK/tools/docs'
```

julianoes, 2025-01-05:
I think it doesn't find the correct OpenSSL in general. This is on my Ubuntu 24.04, with SUPERBUILD:

```
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "3.0.13") 
```

JonasVautherin, 2025-01-05:
Which version of CMake do you have with Ubuntu 24.04?

julianoes, 2025-01-05:
```
cmake --version
cmake version 3.28.3
```

I don't think it's that. I think it's that we're not actually using `find_package(OpenSSL REQUIRED)` if we don't build the mavsdk_server.

Just testing that now, will make a PR soon.

julianoes, 2025-01-05:
Hm, adding the manual `find_package` does not work on 22.04 with cmake 3.22.1. :thinking:

julianoes, 2025-01-05:
Maybe we should throw out openssl and use boringssl instead.

julianoes, 2025-01-05:
Adding this for debugging in src/CMakeLists.txt:

```
set(OPENSSL_ROOT_DIR "${CMAKE_PREFIX_PATH}")
set(OPENSSL_INCLUDE_DIR "${CMAKE_PREFIX_PATH}/include")
set(OPENSSL_LIBRARY_DIR "${CMAKE_PREFIX_PATH}/lib64")

find_package(OpenSSL REQUIRED)

message(STATUS "OPENSSL_FOUND: ${OPENSSL_FOUND}")
message(STATUS "OPENSSL_INCLUDE_DIR: ${OPENSSL_INCLUDE_DIR}")
message(STATUS "OPENSSL_CRYPTO_LIBRARY: ${OPENSSL_CRYPTO_LIBRARY}")
message(STATUS "OPENSSL_CRYPTO_LIBRARIES: ${OPENSSL_CRYPTO_LIBRARIES}")
message(STATUS "OPENSSL_SSL_LIBRARY: ${OPENSSL_SSL_LIBRARY}")
message(STATUS "OPENSSL_SSL_LIBRARIES: ${OPENSSL_SSL_LIBRARIES}")
message(STATUS "OPENSSL_LIBRARIES: ${OPENSSL_LIBRARIES}")
message(STATUS "OPENSSL_VERSION: ${OPENSSL_VERSION}")
message(STATUS "OPENSSL_APPLINK_SOURCE: ${OPENSSL_APPLINK_SOURCE}")
```

Ubuntu 22.04 with cmake 3.22.1:

```
-- OPENSSL_FOUND: TRUE
-- OPENSSL_INCLUDE_DIR: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/include
-- OPENSSL_CRYPTO_LIBRARY: /usr/lib/x86_64-linux-gnu/libcrypto.so
-- OPENSSL_CRYPTO_LIBRARIES: /usr/lib/x86_64-linux-gnu/libcrypto.so
-- OPENSSL_SSL_LIBRARY: /usr/lib/x86_64-linux-gnu/libssl.so
-- OPENSSL_SSL_LIBRARIES: /usr/lib/x86_64-linux-gnu/libssl.so
-- OPENSSL_LIBRARIES: /usr/lib/x86_64-linux-gnu/libssl.so;/usr/lib/x86_64-linux-gnu/libcrypto.so
-- OPENSSL_VERSION: 3.4.0
```

Ubuntu 24.04 with cmake 3.28.3:

```
-- Found OpenSSL: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libcrypto.a   
-- OPENSSL_FOUND: TRUE
-- OPENSSL_INCLUDE_DIR: /home/julianoes/src/upstream/MAVSDK/install-debug;/home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/include
-- OPENSSL_CRYPTO_LIBRARY: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libcrypto.a
-- OPENSSL_CRYPTO_LIBRARIES: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libcrypto.a
-- OPENSSL_SSL_LIBRARY: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libssl.a
-- OPENSSL_SSL_LIBRARIES: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libssl.a
-- OPENSSL_LIBRARIES: /home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libssl.a;/home/julianoes/src/upstream/MAVSDK/build-debug/third_party/install/lib64/libcrypto.a
-- OPENSSL_VERSION: 
-- OPENSSL_APPLINK_SOURCE:
```

What a mess.

julianoes, 2025-01-05:
I managed to copy over the FindOpenSSL.cmake file from 3.28.3 and add it to the modules carried by MAVSDK. This seems to work for `find_package(OpenSSL REQUIRED)` but does not get used within the cURL build. Not sure how to add it there.

JonasVautherin, 2025-01-06:
So CMake 3.28.3 finds it, but CMake 3.22.1 doesn't?

julianoes, 2025-01-06:
@JonasVautherin good idea. Let's see if this works:
https://github.com/mavlink/MAVSDK/pull/2491

PR #2491 (2025-01-10): Find openssl with older cmake version
This copies FindOpenSSL.cmake which comes with cmake 3.28.3 and uses it for the find_package in mavsdk_server as well as curl.

This is required because earlier versions don't work properly which causes linking errors on Ubuntu 22.04.

Should fix #2489.