mavlink/MAVSDK #2493: How can i fix that

dkzdragon02, 2025-01-16:
dkzdragon02@dkzdragon02:~/mavsdk_drone_show$ /bin/python3 /home/dkzdragon02/mavsdk_drone_show/offboard_multiple_from_csv.py
[DEBUG] Simulation Mode: True
[DEBUG] Configuration CSV: config_sitl.csv
[DEBUG] Swarm CSV: swarm_sitl.csv
2025-01-16 11:41:39 [DEBUG] asyncio: Using selector: EpollSelector
/home/dkzdragon02/mavsdk_drone_show/offboard_multiple_from_csv.py:973: DeprecationWarning: connections() is deprecated and will be removed; use net_connections() instead
  for conn in proc.connections(kind='inet'):
2025-01-16 11:41:41 [DEBUG] __main__: Constructed MAVSDK server path: /home/dkzdragon02/.local/lib/python3.10/site-packages/mavsdk/bin/mavsdk_server
2025-01-16 11:41:41 [INFO] __main__: MAVSDK server started with gRPC port 50040 and UDP port 14540.
2025-01-16 11:41:51 [ERROR] __main__: MAVSDK server did not start listening on port 50040 within timeout.
2025-01-16 11:41:51 [ERROR] __main__: Failed to start MAVSDK server. Exiting program.

dkzdragon02, 2025-01-16:
@julianoes

dkzdragon02, 2025-01-16:
https://github.com/alireza787b/mavsdk_drone_show.git   -   this is project, i'm running project on ubutun linux 22.04.5

JonasVautherin, 2025-01-16:
I think that mavsdk_server needs to detect a MAVLink system before it starts the gRPC server.

dkzdragon02, 2025-01-16:
> I think that mavsdk_server needs to detect a MAVLink system before it starts the gRPC server.

let's me try

JonasVautherin, 2025-01-16:
I meant that when you start `mavsdk_server`, it will listen on port `14540` until it receives MAVLink messages from the drone. When it detects the drone, it starts listening on `50051`.

To be confirmed, but if I remember correctly that's how it goes. Which means that you can't just start `mavsdk_server` and immediately try to test for `50051`.

Does that make sense?

dkzdragon02, 2025-01-16:
> I meant that when you start `mavsdk_server`, it will listen on port `14540` until it receives MAVLink messages from the drone. When it detects the drone, it starts listening on `50051`.
>
> To be [...]

let's me try if i can't fix that i will send the message error

julianoes, 2025-03-17:
Did you figure this out?

julianoes, 2025-05-06:
No more response, closing.

closed 2025-05-06 by julianoes