mavlink/MAVSDK #2496: Throw invalid argument exception  in camera plugin

persuader72, 2025-01-18:
This piece of code in camera_definition.cpp at row 745 (in main)

```
`  for (const auto& option : _parameter_map[setting_name]->options) {
        if (option->value == option_name) {
            description = option->name;
            return true;
        }
    }
    LogWarn() << "Option " << option_name << " not found";`
```

Is throwing an  invalid argument exception because **option_name** can't be converted to an unsigned integer.

[image]

Maybe is a typo?

I changed 

```
        if (option->value == option_name) {
```

with

```
        if (option->name == option_name) {
```

and now the library i working normally.

The following is the xml file that is throwing the exception when is parsed.

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<mavlinkcamera>
    <definition version="11">
        <model>GaeboEO</model>
        <vendor>XP4</vendor>
    </definition>
    <parameters>
        <!-- control = 0 tells us this should not create an automatic UI control -->
        <parameter name="CAM_MODE" type="uint32" default="1" control="0">
            <description>Camera Mode</description>
            <options>
                <option name="Photo" value="0"/>
                <option name="Video" value="1"/>
            </options>
        </parameter>
        <parameter name="CAM_ZOOM" type="uint32" default="1" control="1">
            <description>Optical Zoom</description>
            <options>
                <option name="1x" value="1"/>
                <option name="2x" value="2"/>
                <option name="3x" value="3"/>
                <option name="4x" value="4"/>
                <option name="5x" value="5"/>
                <option name="6x" value="6"/>
                <option name="7x" value="7"/>
                <option name="8x" value="8"/>
                <option name="9x" value="9"/>
                <option name="10x" value="10"/>
                <option name="11x" value="11"/>
                <option name="12x" value="12"/>
                <option name="13x" value="13"/>
                <option name="14x" value="14"/>
                <option name="15x" value="15"/>
                <option name="16x" value="16"/>
                <option name="17x" value="17"/>
                <option name="18x" value="18"/>
                <option name="19x" value="19"/>
                <option name="20x" value="20"/>
                <option name="21x" value="21"/>
                <option name="22x" value="22"/>
                <option name="23x" value="23"/>
                <option name="24x" value="24"/>
                <option name="25x" value="25"/>
                <option name="26x" value="26"/>
                <option name="27x" value="27"/>
                <option name="28x" value="28"/>
                <option name="29x" value="29"/>
                <option name="30x" value="30"/>
            </options>
        </parameter>
        <parameter name="CAM_OPTCTRL" type="uint32" writeonly="1" default="3" control="1">
            <description>Camera Zoom</description>
            <options>
                <option name="tele" value="1"/>
                <option name="wide" value="2"/>
                <option name="stop" value="3"/>
            </options>
        </parameter>
        <parameter name="CAM_Z2F" type="float" default="4.3" control="0">
            <description>Zoom to focal length</description>
            <options>
                <option name="1x" value="4.3"/>
                <option name="2x" value="4.4"/>
                <option name="3x" value="4.8"/>
                <option name="4x" value="5.4"/>
                <option name="5x" value="6.1"/>
                <option name="6x" value="6.9"/>
                <option name="7x" value="7.8"/>
                <option name="8x" value="8.0"/>
                <option name="9x" value="10.0"/>
                <option name="10x" value="10.9"/>
                <option name="11x" value="12.7"/>
                <option name="12x" value="14.4"/>
                <option name="13x" value="16.3"/>
                <option name="14x" value="18.6"/>
                <option name="15x" value="21.3"/>
                <option name="16x" value="24.3"/>
                <option name="17x" value="27.6"/>
                <option name="18x" value="31.2"/>
                <option name="19x" value="35.1"/>
                <option name="20x" value="39.5"/>
                <option name="21x" value="44.3"/>
                <option name="22x" value="49.5"/>
                <option name="23x" value="55.6"/>
                <option name="24x" value="62.1"/>
                <option name="25x" value="69.5"/>
                <option name="26x" value="78.2"/>
                <option name="27x" value="88.4"/>
                <option name="28x" value="100.9"/>
                <option name="29x" value="112.1"/>
                <option name="30x" value="126.8"/>
                <option name="31x" value="127.9"/>
                <option name="32x" value="129.0"/>
            </options>
        </parameter>
    </parameters>
</mavlinkcamera>
```

julianoes, 2025-01-19:
Ah, sounds like you're right. Do you mind making a pull request?

persuader72, 2025-01-20:
:ok_hand:, Done in PR [2498](https://github.com/mavlink/MAVSDK/pull/2498)

persuader72, 2025-01-20:
I see now that the PR can't be cherry picked in v2 branch. 
Can I make a different PR for the v2 branch?

julianoes, 2025-01-20:
Sure, you can make a backport PR as well.

julianoes, 2025-03-17:
I assume we can close this, otherwise let me know.

persuader72, 2025-03-17:
I'm sorry for the delay I had some busy days at work,  but give me some more days to analyze this issue, because the problem is still present and my xml parameter file still crash the library in the same point.

As far as i understood until now there is the function CameraDefinition::get_option_str that take the string representation of a camera parameter value and return the correct parameter object.

The problem is that the string representation of the CAM_ZOOM  parameter (for instance) is "**ParamValue{uint32_t:1}**" but this string is converted to an integer by the ParamValue class using a std::stoul function and this throw the exception.

I'm trying to understand why the string  "**ParamValue{uint32_t:1}**" is passed as argument to **CameraDefinition::get_option_str** instead of a simple string containing  "**1**" that will be correctly handled by the ParamValue conversion function..

julianoes, 2025-06-11:
@persuader72 can you test https://github.com/mavlink/MAVSDK/pull/2595?

PR #2595 (2025-06-12): camera: fix invalid argument exception
This should fix a reported invalid argument exception.

Closes #2496.
closed 2025-06-12 by julianoes