mavlink/MAVSDK #2500: Bug in latest versions for Android?

rrajivram, 2025-01-27:
Using mavsdk-2.1.0 and mavsdk-server-2.1.5 and Ardupilot SITL as the backend on an IP port I cannot get telemetry data on Android. Keeps telling me its retrying when I subscribe for position data. The only code I'm using beside server.run() and new System() is this in my main activity.kt : 
   fun runServer() {
        MavsdkEventQueue.executor().execute({
            val address = "tcp://192.168.29.118:5763"
            val port = server.run(address)
            Log.d(TAG, "port is $port")
            if(port == 0) {
                Log.e(TAG,"Unable to start server")
                return@execute
            }
            drone = System("127.0.0.1",port)
            isRunning = true
            buttonText= "Destroy"
            Log.i(TAG,"Server started")
            setupSubscriptions()
        })
    }

    @SuppressLint("CheckResult")
    fun setupSubscriptions() {

        
        val altitude = drone.telemetry.position.subscribeOn(Schedulers.io())
            .observeOn(Schedulers.single())
            .subscribe({
                Log.d(TAG,"info : $it")
                Log.d(TAG,"Lat = " + it.latitudeDeg + ", Lon= " + it.longitudeDeg)
                latText = "   " + it.latitudeDeg
                longText = "   " + it.longitudeDeg
            }, Throwable::printStackTrace)
    }

I've validated that QGC and mavproxy are able to both communicate with the SITL. When I change the mavsdk and server versions to 2.0 , then this works perfectly.

julianoes, 2025-01-27:
mavsdk-server-2.1.5 seems quite old? Any chance you can try latest?

Do you have an example for me to test this with?

rrajivram, 2025-01-28:
2.1.5 is the latest version on Apache maven..

the entire source code is what I posted - just added a button to start the
server that calls start server (posted). I’ll try to post the entire kt
 file but it doesn’t have any new info.

JonasVautherin, 2025-01-28:
Does it detect a mavlink system? Does your drone report receiving messages from MAVSDK over TCP?

rrajivram, 2025-01-28:
Yes. I tried both arm and disarm commands and they worked fine. I wasn’t
able to get any debugging output on ACKs , but I was able to see the SITL
received and acted on the commands. Funnily enough, the SITL also disarmed
automatically and notified my client it did so, even if I didn’t send the
disarm command . I haven’t tried the commands yet with the 1.x versions -
can try that if it will help.

rrajivram, 2025-01-28:
And now its stopped working ; tried both version 1.2 and version 2.0. The entire source code is below : 
`package com.example.greeting

import android.annotation.SuppressLint
import android.os.Bundle
import androidx.activity.ComponentActivity
import androidx.activity.compose.setContent
import androidx.activity.enableEdgeToEdge
import androidx.activity.viewModels
import androidx.compose.foundation.layout.Arrangement
import androidx.compose.foundation.layout.Column
import androidx.compose.foundation.layout.fillMaxSize
import androidx.compose.foundation.layout.padding
import androidx.compose.material3.Surface
import androidx.compose.material3.Text
import androidx.compose.runtime.Composable
import androidx.compose.ui.Alignment
import androidx.compose.ui.Modifier
import androidx.compose.ui.graphics.Color
import androidx.compose.ui.tooling.preview.Preview
import androidx.compose.ui.unit.dp
import com.example.greeting.ui.theme.GreetingTheme
import android.util.Log
import androidx.compose.foundation.layout.Row
import androidx.compose.material3.Button
import androidx.compose.runtime.getValue
import androidx.compose.runtime.mutableStateOf
import androidx.compose.runtime.remember
import androidx.compose.runtime.setValue
import io.mavsdk.MavsdkEventQueue;
import io.mavsdk.System;
import io.mavsdk.mavsdkserver.MavsdkServer;
import io.mavsdk.telemetry.Telemetry
import io.mavsdk.telemetry.Telemetry.Position
import io.reactivex.disposables.Disposable
import io.reactivex.Flowable;
import io.reactivex.functions.Consumer
import io.reactivex.functions.Function;
import io.reactivex.schedulers.Schedulers;
import kotlinx.coroutines.flow.Flow

class MainActivity : ComponentActivity() {
   private val TAG = "MainActivity"

    private val viewModel by viewModels<DroneModel>()
    private val server = MavsdkServer()
    private lateinit var drone: System
    private var isRunning = false
    private var disposables = mutableListOf<Disposable>()

    var buttonText by  mutableStateOf("Start")
    var altitudeText by mutableStateOf("AL")
    var latText by mutableStateOf("LA")
    var longText by mutableStateOf("LO")

    @SuppressLint("QueryPermissionsNeeded")
    override fun onCreate(savedInstanceState: Bundle?) {
        java.lang.System.setProperty("org.slf4j.simpleLogger.defaultLogLevel","Trace")
        Log.d(TAG,"in on create")
        super.onCreate(savedInstanceState)
        enableEdgeToEdge()
        setContent {
            GreetingTheme {
                Column(
                    modifier = Modifier.fillMaxSize(),
                    horizontalAlignment = Alignment.CenterHorizontally,
                    verticalArrangement = Arrangement.Center
                ) {
                    Row() {
                        Button(onClick = {
[... 128 lines cut ...]

Here are the log messages I'm seeing:
2025-01-28 10:45:58.089  6832-6892  Mavsdk                  com.example.greeting                 I  MAVSDK version: v2.12.2
2025-01-28 10:45:58.090  6832-6892  MAVSDK-Server           com.example.greeting                 D  Running mavsdk_server with connection url: tcp://192.168.29.118:5763
2025-01-28 10:45:58.090  6832-6892  Mavsdk                  com.example.greeting                 I  Waiting to discover system on tcp://192.168.29.118:5763...
2025-01-28 10:45:58.107  6832-6896  Mavsdk                  com.example.greeting                 D  New system ID: 1 Comp ID: 1
2025-01-28 10:45:58.107  6832-6896  Mavsdk                  com.example.greeting                 D  Component Autopilot (1) added.
2025-01-28 10:45:58.107  6832-6896  Mavsdk                  com.example.greeting                 W  Vehicle type changed (new type: 2, old type: 0)
2025-01-28 10:45:58.107  6832-6896  Mavsdk                  com.example.greeting                 D  Discovered 1 component(s)
2025-01-28 10:45:58.108  6832-6894  Mavsdk                  com.example.greeting                 I  System discovered
2025-01-28 10:45:58.115  6832-6892  Mavsdk                  com.example.greeting                 I  Server started
2025-01-28 10:45:58.115  6832-6892  Mavsdk                  com.example.greeting                 I  Server set to listen on 0.0.0.0:35297
2025-01-28 10:45:58.115  6832-6892  MAVSDK-Server           com.example.greeting                 D  mavsdk_server is now running, listening on port 35297
2025-01-28 10:45:58.113  6832-6832  mavsdk-event-qu         com.example.greeting                 W  type=1400 audit(0.0:3935): avc:  denied  { read } for  name="somaxconn" dev="proc" ino=496700 scontext=u:r:untrusted_app:s0:c190,c256,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0 app=com.example.greeting
2025-01-28 10:45:58.115  6832-6892  MainActivity            com.example.greeting                 D  port is 35297
2025-01-28 10:45:58.118  6832-6892  MainActivity            com.example.greeting                 I  Server started
2025-01-28 10:45:58.128  6832-6892  System.err              com.example.greeting                 W  SLF4J(W): No SLF4J providers were found.
2025-01-28 10:45:58.128  6832-6892  System.err              com.example.greeting                 W  SLF4J(W): Defaulting to no-operation (NOP) logger implementation
2025-01-28 10:45:58.128  6832-6892  System.err              com.example.greeting                 W  SLF4J(W): See https://www.slf4j.org/codes.html#noProviders for further details.
[... cut ...]
2025-01-28 10:45:58.410  6832-6896  Mavsdk                  com.example.greeting                 D  Component Unsupported component (230) added.
2025-01-28 10:45:58.410  6832-6896  Mavsdk                  com.example.greeting                 D  Discovered 1 component(s)
2025-01-28 10:45:59.429  6832-6921  ProfileInstaller        com.example.greeting                 D  Installing profile for com.example.greeting
2025-01-28 10:46:07.967  6832-6877  EGL_emulation           com.example.greeting                 D  app_time_stats: avg=202.23ms min=1.90ms max=9099.12ms count=49

JonasVautherin, 2025-01-28:
> Yes. I tried both arm and disarm commands and they worked fine.

This sounds like mavlink messages sent by MAVSDK are received by the autopilot, but mavlink messages sent by the autopilot are not received by MAVSDK. Because you use a TCP connection, I don't think it would be a problem there (usually that happens with UDP) 🤔.

Are you sure that you don't have multiple clients competing on the connection? Not sure how Ardupilot implements TCP support. Did you make your tests when only MAVSDK and Ardupilot are running, and not pymavlink or QGC in parallel?

rrajivram, 2025-01-28:
I'm sure about multiple clients. I do have mavproxy running (as part of
sim_vehicle), but that is only listening on localhost on UDP.  I confirmed
in the SITL run that only one connection is active.

PabloAsensio, 2025-02-20:
I’m encountering the same issue! I’m currently migrating a Python DroneKit app from Windows to Android, and it seems there’s a problem with the TCP protocol. Specifically, it appears that the Android app isn’t sending the required message to properly configure the connection.

When I test the DroneKit version using TCP (e.g., port 5760), it works fine. Similarly, the MAVSDK version operates as expected. However, when I attempt to use the Android app without first running the Windows version, the app fails to listen for the connection unless the link is already established.

It seems that we need to ensure the connection is properly established before attempting to listen with the Android app.

This doesn’t seem to be an SITL issue, since I can successfully listen to telemetry on Mission Planner, QGC, and their respective Android versions. I’m 100% sure the problem is related to the TCP protocol.

PabloAsensio, 2025-02-21:
If hepls. Catlogs differences `2.1.5` vs `2.1.6`  (using same code):

-  `2.1.5` Many Mavsdk warnings and nothing recieved (some traces in spanish like, start, stop...)                                 

```
2025-02-21 12:33:18.698  3237-3637  Mavsdk                                  I  MAVSDK version: v2.14.0
2025-02-21 12:33:18.699  3237-3637  MAVSDK-Server                           D  Running mavsdk_server with connection url: tcp://10.0.2.2:5762
2025-02-21 12:33:18.699  3237-3637  Mavsdk                                  I  Waiting to discover system on tcp://10.0.2.2:5762...
2025-02-21 12:33:18.805  3237-5094  Mavsdk                                  D  New system ID: 1 Comp ID: 1
2025-02-21 12:33:18.806  3237-5094  Mavsdk                                  D  Component Autopilot (1) added.
2025-02-21 12:33:18.806  3237-5094  Mavsdk                                  W  Vehicle type changed (new type: 2, old type: 0)
2025-02-21 12:33:18.806  3237-5094  Mavsdk                                  D  Discovered 1 component(s)
2025-02-21 12:33:18.849  3237-3637  Mavsdk                                  I  System discovered
2025-02-21 12:33:19.274  3237-3408  EGL_emulation                           D  app_time_stats: avg=13221.21ms min=29.35ms max=26413.07ms count=2
2025-02-21 12:33:19.309  3237-3237  DefaultDispatch                         W  type=1400 audit(0.0:93): avc:  denied  { read } for  name="somaxconn" dev="proc" ino=58960 scontext=u:r:untrusted_app:s0:c195,c256,c512,c768 tcontext=u:object_r:proc_net:s0 tclass=file permissive=0 app=
2025-02-21 12:33:19.341  3237-3637  Mavsdk                                  I  Server started
2025-02-21 12:33:19.341  3237-3637  Mavsdk                                  I  Server set to listen on 0.0.0.0:32897
2025-02-21 12:33:19.341  3237-3637  MAVSDK-Server                           D  mavsdk_server is now running, listening on port 32897
2025-02-21 12:33:19.341  3237-3637  PixhawkHandler                          D  Servidor MAVSDK iniciado en el puerto: 32897
2025-02-21 12:33:19.436  3237-3637  PixhawkHandler                          D  Servidor iniciado en puerto: 32897
2025-02-21 12:33:20.175  3237-5155  TrafficStats                            D  tagSocket(101) with statsTag=0xffffffff, statsUid=-1
2025-02-21 12:33:20.494  3237-5155  TrafficStats                            D  tagSocket(106) with statsTag=0xffffffff, statsUid=-1
2025-02-21 12:33:21.146  3237-5091  Mavsdk                                  W  sending again, retries to do: 5
2025-02-21 12:33:21.651  3237-5091  Mavsdk                                  W  sending again, retries to do: 4
2025-02-21 12:33:22.156  3237-5091  Mavsdk                                  W  sending again, retries to do: 3
2025-02-21 12:33:22.670  3237-5091  Mavsdk                                  W  sending again, retries to do: 2
2025-02-21 12:33:23.179  3237-5091  Mavsdk                                  W  sending again, retries to do: 1
2025-02-21 12:33:23.688  3237-5091  Mavsdk                                  E  retrying failed
2025-02-21 12:33:23.688  3237-5091  Mavsdk                                  E  Error: Param for gyro offset_x failed.
2025-02-21 12:33:24.216  3237-5091  Mavsdk                                  W  sending again, retries to do: 5
2025-02-21 12:33:24.719  3237-5091  Mavsdk                                  W  sending again, retries to do: 4
2025-02-21 12:33:25.235  3237-5091  Mavsdk                                  W  sending again, retries to do: 3
2025-02-21 12:33:25.736  3237-5091  Mavsdk                                  W  sending again, retries to do: 2
2025-02-21 12:33:26.241  3237-5091  Mavsdk                                  W  sending again, retries to do: 1
2025-02-21 12:33:26.744  3237-5091  Mavsdk                                  E  retrying failed
2025-02-21 12:33:26.744  3237-5091  Mavsdk                                  E  Error: Param for gyro offset_y failed.
2025-02-21 12:33:27.269  3237-5091  Mavsdk                                  W  sending again, retries to do: 5
2025-02-21 12:33:27.778  3237-5091  Mavsdk                                  W  sending again, retries to do: 4
2025-02-21 12:33:28.289  3237-5091  Mavsdk                                  W  sending again, retries to do: 3
2025-02-21 12:33:28.790  3237-5091  Mavsdk                                  W  sending again, retries to do: 2
2025-02-21 12:33:29.301  3237-5091  Mavsdk                                  W  sending again, retries to do: 1
2025-02-21 12:33:29.806  3237-5091  Mavsdk                                  E  retrying failed
2025-02-21 12:33:29.806  3237-5091  Mavsdk                                  E  Error: Param for gyro offset_z failed.
2025-02-21 12:33:30.328  3237-5091  Mavsdk                                  W  sending again, retries to do: 5
2025-02-21 12:33:30.840  3237-5091  Mavsdk                                  W  sending again, retries to do: 4
2025-02-21 12:33:31.343  3237-5091  Mavsdk                                  W  sending again, retries to do: 3
2025-02-21 12:33:31.849  3237-5091  Mavsdk                                  W  sending again, retries to do: 2
2025-02-21 12:33:32.360  3237-5091  Mavsdk                                  W  sending again, retries to do: 1
2025-02-21 12:33:32.870  3237-5091  Mavsdk                                  E  retrying failed
2025-02-21 12:33:32.870  3237-5091  Mavsdk                                  E  Error: Param for accel offset_x failed.
[... cut ...]
2025-02-21 13:06:54.207  6745-6745  DEBUG                   crash_dump64                         A        #63 pc 000000000000cef8  <anonymous:7340b147e000> (.pixhawk.PixhawkHandler.access$cleanUpResources+0)
2025-02-21 13:06:54.207  6745-6745  DEBUG                   crash_dump64                         A        #69 pc 000000000000c474  <anonymous:7340b147e000> (.pixhawk.PixhawkHandler$destroyMavsdkServer$1.invokeSuspend+0)
---------------------------- PROCESS ENDED (6641) for package  ----------------------------
---------------------------- PROCESS ENDED (6748) for package  ----------------------------
```

PabloAsensio, 2025-02-21:
tcp error can be seen on the 2.1.5

julianoes, 2025-05-06:
Can you provide a small example app that would allow us to reproduce this issue?