mavlink/MAVSDK #2505: Offboard::VelocityBodyYawspeed not responding

samirertt, 2025-02-10:
sleep_for(seconds(10));

    auto offboard = Offboard{system};
    Offboard::VelocityBodyYawspeed stay{};
    offboard.set_velocity_body(stay);

    Offboard::Result offboard_result = offboard.start();
    if (offboard_result != Offboard::Result::Success) {
        std::cerr << "Offboard modu baslatilamadi: " << offboard_result << std::endl;
        return 1;
    }
    Telemetry::Position position = telemetry.position();
    std::cout << "Offboard modu baslatildi. Saga donud yapiliyor..." << std::endl;

    
    Offboard::VelocityBodyYawspeed saga_donus{10.0f, 5.0f, 0.0f, 0.0f};
    std::cout << "Position: longitute: " << position.longitude_deg << " latitude: " << position.latitude_deg << std::endl;
    offboard.set_velocity_body(saga_donus); 
    sleep_for(seconds(5)); 

    std::cout << "Şimdi sola dönüş yapılıyor..." << std::endl;

    Offboard::VelocityBodyYawspeed sola_donus{5.0f, -10.0f, 0.0f, 0.0f};

    offboard.set_velocity_body(sola_donus);
    sleep_for(seconds(5)); 

    
    Offboard::VelocityBodyYawspeed yukari_hareket{5.0f, 0.0f, -5.0f, 0.0f};
    std::cout << "Yukarı hareket ediliyor..." << std::endl;

    offboard.set_velocity_body(yukari_hareket);
    sleep_for(seconds(5)); 

    
    Offboard::VelocityBodyYawspeed asagi_hareket{5.0f, 0.0f, 10.0f, 0.0f};
    std::cout << "Aşağı hareket ediliyor..." << std::endl;

    offboard.set_velocity_body(asagi_hareket);
    sleep_for(seconds(5));

    
    offboard_result = offboard.stop();
    if (offboard_result != Offboard::Result::Success) {
        std::cerr << "Offboard modu durdurulamadı: " << offboard_result << std::endl;
        return 1;
    }

    std::cout << "Offboard modu durduruldu." << std::endl;

this is my offboard mode part. its all fine before and after no errors. 

help me out guys

julianoes, 2025-02-12:
Is this PX4 or ArduPilot?

samirertt, 2025-02-12:
it is PX4 and i have solved the problem thanks for asking i had to use attitude  and not VelocityBodyYawspeed