mavlink/MAVSDK #2507: Unable to control the drone with the manual control plugin even though .setManualControlInput() function return success . Does MavSDK implement sending RC_OVERRIDE messages when using the manual control plugin?

Ba0Nguyen, 2025-02-12:
I am able to use the manual control plugin to control the drone in SITL. However, I am unable to control my physical drone. I'm thinking it might be related to not overriding the RC Radio on the Flight Controller. One of the error that I saw before arming the drone is Prearm: RC not found.
Does MavSDK implement sending RC_OVERRIDE messages when using the manual control plugin?

julianoes, 2025-02-12:
Is this using PX4 or ArduPilot?

Ba0Nguyen, 2025-02-12:
This is using Ardupilot.

julianoes, 2025-02-12:
Can you check the param `SYSID_MYGCS`?
https://ardupilot.org/copter/docs/parameters-Copter-stable-V4.3.2.html#sysid-mygcs-my-ground-station-number

Ba0Nguyen, 2025-02-12:
SYSID_MYGCS is 255

Ba0Nguyen, 2025-02-13:
I think this might be the issue, I will test it with my drone.

Ba0Nguyen, 2025-02-13:
@julianoes , I tries with SysID 255 and it works!

closed 2025-02-13 by Ba0Nguyen