mavlink/MAVSDK #2510: rtcm messages unidentical with mavsdk::rtk.h

Wenz922, 2025-02-12:
I am using mavsdk::rtk.h to send rtcm messages to a Pixhawk 6C. Here is a part of code: 

`void subscribe_to_rtcm() {
        rtcm_subscription_ = this->create_subscription<std_msgs::msg::String>(
            rtcm_topic_, 10,
            [this](const std_msgs::msg::String::SharedPtr msg) {
                forward_to_pixhawk(msg->data);
            });
    }

    void forward_to_pixhawk(const std::string& rtcm_data_str) {
        mavsdk::Rtk::RtcmData rtcm_data;
        rtcm_data.data = rtcm_data_str;

        // Send RTCM data using the RTK plugin
        Rtk::Result result = rtk_plugin_->send_rtcm_data(rtcm_data);
        if (result == Rtk::Result::Success) {
            RCLCPP_INFO(this->get_logger(), "RTCM data forwarded to Pixhawk: %zu bytes", rtcm_data_str.size());
        } else {
            RCLCPP_WARN(this->get_logger(), "Failed to forward RTCM data: %d", static_cast<int>(result));
        }
    }`
But in logging the rtcm data size is not identical, What is the problem?
[INFO] [1739119220.793310883] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119220.893626373] [rtcm_base_node]: Published RTCM data of length: 27 bytes
[INFO] [1739119221.075379546] [rtcm_base_node]: Published RTCM data of length: 144 bytes
[INFO] [1739119221.093328974] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.193289930] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.293425000] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.393339313] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.493329355] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.593762593] [rtcm_base_node]: Published RTCM data of length: 256 bytes
[INFO] [1739119221.693307979] [rtcm_base_node]: Published RTCM data of length: 256 bytes

[INFO] [1739119210.293539628] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 256 bytes
[INFO] [1739119210.393929978] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 256 bytes
[INFO] [1739119210.493808059] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 168 bytes
[INFO] [1739119210.593585514] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 57 bytes
[INFO] [1739119210.693704616] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 15 bytes
[INFO] [1739119210.793863885] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 101 bytes
[INFO] [1739119210.893734223] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 1 bytes
[INFO] [1739119210.993963822] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 44 bytes
[INFO] [1739119211.093535019] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 0 bytes
[INFO] [1739119211.193466552] [rtcm_rover_node]: RTCM data forwarded to Pixhawk: 256 bytes

reedev, 2025-02-12:
Hi, based on this small snippet (so no context): do you realize that rtcm is a stream of data? As long as the whole stream contains the same content, it is ok. Packets are sent in chunks.

Have you checked the whole stream for missing bytes?

Please correct me if I am wrong but this is my understanding at the moment.

julianoes, 2025-02-12:
@Wenz922 are you using v2 or v3? With v3 the rtcm data needs to be supplied base64 encoded.

Wenz922, 2025-02-12:
@julianoes i am using v2.12.2. 
According to the MAVSDK API rtk.h documentation, RTCM data should be sent as raw binary (std::string), not base64 encoded or?
Btw. i tried before base64 encoded to verify the whole rtcm message with CRC while using MavlinkPassthrough, and it had similar problem.

julianoes, 2025-02-13:
Correct, no base64 with v2.

And so are corrections not working at all? Can you share any other information about your setup? E.g. PX4 or ArduPilot? And what versions?

Wenz922, 2025-02-13:
@julianoes There are only partial RTCM messages that are received, so that the correction doesn't work. I am using PX4 version1.15.2.
Here is main part code: thank you a lot ;)(the layout might look strange, i tried my best..) 
class RTCMMAVSDKToPixhawk : public rclcpp::Node {
public:
    RTCMMAVSDKToPixhawk()
        : Node("rtcm_fcu_mavsdk_node") {
        // Declare and get parameters
        this->declare_parameter<std::string>("serial_port", "/dev/ttyAMA2");
        this->declare_parameter<int>("baud_rate", 57600);
        this->declare_parameter<std::string>("rtcm_topic", "/rtcm_data");`

        serial_port_ = this->get_parameter("serial_port").as_string();
        baud_rate_ = this->get_parameter("baud_rate").as_int();
        rtcm_topic_ = this->get_parameter("rtcm_topic").as_string();

        // Initialize MAVSDK
        setup_mavsdk();
    }

private:
    void setup_mavsdk() {
        // Initialize MAVSDK with Companion Computer configuration
        mavsdk_ = std::make_shared<Mavsdk>(Mavsdk::Configuration{Mavsdk::ComponentType::CompanionComputer});
        ConnectionResult connection_result = mavsdk_->add_serial_connection(serial_port_, baud_rate_);

        if (connection_result != ConnectionResult::Success) {
            RCLCPP_ERROR(this->get_logger(), "Failed to connect to Pixhawk: %s", connection_result_to_string(connection_result).c_str());
            schedule_mavsdk_reconnection();
            return;
        }

        RCLCPP_INFO(this->get_logger(), "Connecting to Pixhawk...");
        wait_for_system();
    }

    void wait_for_system() {
        // Wait for the system to connect
        while (mavsdk_->systems().empty()) {
            RCLCPP_INFO(this->get_logger(), "Waiting for Pixhawk system...");
            std::this_thread::sleep_for(std::chrono::seconds(1));
        }

        // Get connected system
        system_ = mavsdk_->systems().at(0);
        rtk_plugin_ = std::make_shared<Rtk>(system_);

        if (!rtk_plugin_) {
            RCLCPP_ERROR(this->get_logger(), "Failed to initialize RTK plugin.");
            schedule_mavsdk_reconnection();
            return;
        }

        RCLCPP_INFO(this->get_logger(), "Pixhawk system connected. RTK plugin initialized.");

        // Subscribe to RTCM topic
        subscribe_to_rtcm();
    }

    void subscribe_to_rtcm() {
        rtcm_subscription_ = this->create_subscription<std_msgs::msg::String>(
            rtcm_topic_, 10,
            [this](const std_msgs::msg::String::SharedPtr msg) {
                forward_to_pixhawk(msg->data);
            });
    }

    void forward_to_pixhawk(const std::string& rtcm_data_str) {
        mavsdk::Rtk::RtcmData rtcm_data;
        rtcm_data.data = rtcm_data_str;

        // Send RTCM data using the RTK plugin
        Rtk::Result result = rtk_plugin_->send_rtcm_data(rtcm_data);
        if (result == Rtk::Result::Success) {
            RCLCPP_INFO(this->get_logger(), "RTCM data forwarded to Pixhawk: %zu bytes", rtcm_data_str.size());
        } else {
            RCLCPP_WARN(this->get_logger(), "Failed to forward RTCM data: %d", static_cast<int>(result));
        }
    }

    void schedule_mavsdk_reconnection() {
    if (!mavsdk_reconnect_timer_) {
        mavsdk_reconnect_timer_ = this->create_wall_timer(
            std::chrono::seconds(5), // Explicitly specify duration as 5 seconds
            [this]() {
                RCLCPP_INFO(this->get_logger(), "Attempting to reconnect to Pixhawk...");
                setup_mavsdk();
            });
        }
    }
`

Wenz922, 2025-02-16:
Hi @julianoes , i have checked the version of MAVSDK and PX4 which i used. And the RTK devices are two H-RTK ZED-F9P Helical, one as Base and another one as Rover. Configuration is as followed : https://docs.holybro.com/gps-and-rtk-system/zed-f9p-h-rtk-series/portable-rtk-base-station-setup. It works well directly with holybro telemtry radio. Is it probably the reason of holybro telemtry with baudrate 57600?

julianoes, 2025-02-16:
I will have to create a C++ example and test that myself.

Wenz922, 2025-02-16:
@julianoes that will be great! thanks a lot, stay updated ;)

reedev, 2025-02-16:
Hi, fwiw, I will double check my setup with MAVSDK v3.0.0 since I have used my setup successfully with MAVSDK v2

Wenz922, 2025-02-16:
@reedev would you mind to share more detail, for example the code part of library rtk.h?

reedev, 2025-02-17:
As I said, I will check MAVSDK v3 with rtk, to see if it still works as expected as with v2. We know that v2 was working ok so I suggest to make your setup with v2 first.

I don't see that you tried your setup with a different setup, are you sure your equipment, server etc are working properly?

reedev, 2025-02-17:
Aha sorry, I missed that part of your story. In that case I can't help you much further since my equipment and setup is completely different. I will test MAVSDK v3 to make sure that the MAVSDK is working ok.

Wenz922, 2025-02-19:
Hi @julianoes sorry to bother you, do you have any updates about this issue? Or are there any alternative solutions?

Wenz922, 2025-02-19:
@reedev no problem, would you mind to share your setup? Probably i could learn from it.

julianoes, 2025-02-19:
It's on my list to recreate the setup.

reedev, 2025-02-20:
whenever I have more details to share, I will get back.

Wenz922, 2025-02-21:
i changed the code of publisher and subscriber a bit, now the RTCM messages are being successfully published and received identically by the Pixhawk but still i am not getting an RTK FIX status.
Here is the publisher:
`void read_and_publish_rtcm() {
        std::vector<uint8_t> buffer(256); // Create a buffer of 256 bytes
        try {
            size_t bytes_read = serial_conn_.read_some(boost::asio::buffer(buffer, buffer.size()));

            // Resize vector to contain only received data
            buffer.resize(bytes_read);

            // Convert to HEX string for logging
            std::stringstream hex_stream;
            hex_stream << std::hex << std::setfill('0');  // Ensure proper formatting

            for (uint8_t byte : buffer) {
              hex_stream << std::setw(2) << static_cast<int>(byte) << " ";
            }

            // Log RTCM message in HEX format
            RCLCPP_INFO(this->get_logger(), "RTCM Data Size: %lu bytes", bytes_read);
            //RCLCPP_INFO(this->get_logger(), "RTCM Data (Hex): %s", hex_stream.str().c_str());

            // Publish RTCM message
            std_msgs::msg::UInt8MultiArray msg;
            msg.data.assign(buffer.begin(), buffer.end());
            
            rtcm_pub_->publish(msg);
        } catch (std::exception &e) {
            RCLCPP_ERROR(this->get_logger(), "Error reading serial data: %s", e.what());
        }
    }`

Here is the subscriber:
`void subscribe_to_rtcm() {
        rtcm_subscription_ = this->create_subscription<std_msgs::msg::UInt8MultiArray>(
            "rtcm_data", 10,
            [this](const std_msgs::msg::UInt8MultiArray::SharedPtr msg) {
                forward_to_pixhawk(std::vector<uint8_t>(msg->data.begin(), msg->data.end()));
            });
    }    

    void forward_to_pixhawk(const std::vector<uint8_t>& rtcm_data) {
        if (!rtk_plugin_) {
            RCLCPP_ERROR(this->get_logger(), "RTK Plugin not initialized!");
            return;
        }
    
        mavsdk::Rtk::RtcmData mav_rtcm_data;
        
        // Convert std::vector<uint8_t> to std::string
        mav_rtcm_data.data.assign(rtcm_data.begin(), rtcm_data.end());
    
        // Debugging: Print RTCM Data in HEX
        std::ostringstream oss;
        for (uint8_t byte : rtcm_data) {
            oss << std::hex << std::setw(2) << std::setfill('0') << (int)byte << " ";
        }
        //RCLCPP_INFO(this->get_logger(), "RTCM Data (Hex): %s", oss.str().c_str());
    
        mavsdk::Rtk::Result result = rtk_plugin_->send_rtcm_data(mav_rtcm_data);
        if (result == mavsdk::Rtk::Result::Success) {
            RCLCPP_INFO(this->get_logger(), "RTCM Data Size: %lu bytes", rtcm_data.size());
            RCLCPP_INFO(this->get_logger(), "Successfully sent RTCM data to Pixhawk");
        } else {
            RCLCPP_WARN(this->get_logger(), "Failed to forward RTCM data: %d", static_cast<int>(result));
        }
    }`

aminballoon, 2025-03-13:
@Wenz922 Cloud you please share 
`$ listener gps_inject_data` in PX4 NuttX shell for ensure RTCM was come in PX4 in case MAVSDK and Holybro telemetry

julianoes, 2025-03-18:
I found my old example and updated it for MAVSDK v3, see if this works for you: https://github.com/julianoes/rtk-sender-example

I haven't recently tested it but will do so later today.

Wenz922, 2025-03-18:
@julianoes Ok, thanks very much for your confirm that it is not the problem of code. Could you share the detail of setting?
I realize since last week that it could be the problem of setting of RTK. I have the RTK F9P Helical (base and rover) and tested firstly with QGroundControl with default configuration, they got also successfully into RTK fixed (https://docs.px4.io/main/en/gps_compass/rtk_gps.html), I could find the parameters RTK_INJECT_GPS1 and gps_inject_data in QGC. But since i manually configured the base and rover with u-center (https://docs.holybro.com/gps-and-rtk-system/zed-f9p-h-rtk-series/portable-rtk-base-station-setup), these parameters were disappeared in QGC and they couldn't get in RTK fixed @aminballoon . Does anyone have more information? Thank you in advance!

julianoes, 2025-03-19:
I don't know my configuration. I didn't mess with that.

reedev, 2025-03-19:
Fwiw, I tested with MAVSDK v3 and after I changed the rtcm data into base64, everything was still working as before. I don't have a special setup, just using rtcm data from some ntrip server, convert it to base64 and send it via the rtk plugin.

I have no special params set.

Wenz922, 2025-03-21:
Hi @julianoes it works greatly with RTK Fix while i use default configuration, thanks a lot anyway!

julianoes, 2025-03-27:
Ok great. I assume we can close this issue then.

closed 2025-03-27 by julianoes