mavlink/MAVSDK #2516: Telemetry rates: set_rate_rc unsupported

matthieu-db, 2025-02-27:
On Ardupilot, I need to set the telemetry rates myself or I don't get any updates. This works for battery and GPS, for instance.

set_rate_rc is not implemented, I guess because it is already sent with system status. However I don't get any updates there unless I open QGroundControl and then come back to my app.

I guess this is more of a list of questions:

- How does a GCS make the rc connection status start working?
- How do I get the connection strength percentage? It is always NaN
- I don't really want to change the rates that are configured in parameters on the autopilot. Is there a way to set the rate to the value configured on the autopilot? Or do I need to fetch the parameter and then set it to that value with set_rate_* ? In that case, how do I know which parameter to check (i.e. which telemetry port: for ArduPilot x in SRx_*)?

matthieu-db, 2025-02-28:
I'm now getting  `COMMAND_DENIED: Command Denied` with telemetry.setRateAltitude(1.0)

StatusText from telemetry has:

> INFO No ap_message for mavlink id (141)

No clue what this means?

Again, working fine with PX4

JonasVautherin, 2025-02-28:
Thanks for looking into that! Always happy to get contributions improving Ardupilot support 😊

matthieu-db, 2025-02-28:
Haha that was not the answer I was hoping for but fair enough, I'll try to figure it out and maybe contribute something back :)

JonasVautherin, 2025-02-28:
Did you try asking on the Ardupilot Discord? I know that some people are using MAVSDK with Ardupilot (they contributed the support we have now) and they may be able to help you. They probably won't see this issue here, so it's probably worth linking it on their Discord 👍.

matthieu-db, 2025-02-28:
- For the altitude, I got it from telemetry.position
- For the speed, the ODOMETRY message is also not supported by ArduPilot. Instead I got it from telemetry.velocityNed - I just needed the sqrt of the three components so that should be fine too
- still outstanding for me: the RC connection status question

@JonasVautherin I will try the discord, thanks for the suggestion!

matthieu-db, 2025-05-28:
After https://github.com/mavlink/MAVSDK/commit/e2b815e29d2cf6bc6823aa9d3d02d3dc0019123a, there is a workaround by just calling set_rate_health (which will also set SYS_UPDATE).

I think we should have a set_rate_sys_status and get rid of the health and rc_status ones. Or just have both do the same thing but that could be confusing for users.

Surya-prathamzzz, 2025-10-01:
hey @matthieu-db you got any lead regarding this!???

matthieu-db, 2025-10-01:
> hey [@matthieu-db](https://github.com/matthieu-db) you got any lead regarding this!???

Did you read the discussion above? Call set_rate_health