mavlink/MAVSDK #2522: RC_Status and mavlink remote health

aminballoon, 2025-03-14:
MAVSDK `RcStatus` will return `is_available = False` while autopilot has can't be armed

[PX4 manualControlCheck.cpp](https://github.com/aminballoon/PX4-Autopilot/blob/v1.15.0-alpha1/src/modules/commander/HealthAndArmingChecks/checks/manualControlCheck.cpp)

In this case if **Kill switch engaged** *RCStatus::is_available will be False** by this Flag

[RC_OK Flag check](https://github.com/mavlink/MAVSDK/blob/2304566ad9d1b7aa98bf0f12baee5f7151613589/src/mavsdk/plugins/telemetry/telemetry_impl.cpp#L1029-L1030C87) ``` const bool rc_ok = sys_status.onboard_control_sensors_health & MAV_SYS_STATUS_SENSOR_RC_RECEIVER; ```

In my opinion, when the kill switch is engaged, it doesn't mean that the RC is not available; it's available, but it just can't be armed.

The `rc_ok` Flag should not be check only MAV_SYS_STATUS_SENSOR_RC_RECEIVER or I need to contribute manualControlCheck in PX4 . 

I can add proto [Mavlink RC_CHANNELS_RAW](https://mavlink.io/en/messages/common.html#:~:text=UINT8_MAX%3A%20invalid/unknown.-,RC_CHANNELS_RAW,-(35)) for check the latest RC input by Timestamp and solve this issue.
What do you think about this issue and my solution ?

julianoes, 2025-03-14:
Interesting. I would argue this needs changing/fixing on the PX4 side.

julianoes, 2025-06-02:
Sounds like a PX4 issue, so I'm closing this.

closed 2025-06-02 by julianoes