mavlink/MAVSDK #2529: CMAKE parse errors when configuring on Windows x64

luterofromtx, 2025-03-24:
Windows 10 Pro x64

Full disclosure, I am a student and new to CMAKE as well as working with third party C++ libraries, so I hope I will have provided enough information here. On my first attempt building from source, I initially ran into an error telling me that openssl "is not compatible with my version of Windows", yet when installing an setting to %path% both the 32 bit version, verifying it works with "openssl -v", then repeating the process with the 64 bit version, I was able to get both running just fine outside of working with this repo. I left the 64 bit version installed and tried again by starting over from the beginning, and now I don't even reach that stage. When trying to build again, I receive the following error message. 

CMake Error at liblzma-stamp/download-liblzma.cmake:167:
  Parse error.  Function missing ending ")".  Instead found unterminated
  string with text "

  )

  ".

CMake Error: Error processing file: C:/Users/logst/MAVSDK/build/default/third_party/liblzma/liblzma-prefix/src/liblzma-stamp/download-liblzma.cmake
ninja: build stopped: subcommand failed.
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  C:/Users/logst/MAVSDK/build/default/third_party/liblzma failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:14 (build_target)

-- Configuring incomplete, errors occurred!

I looked in the local repository and could not find the file specified, so I am not sure what else to do here. My whole purpose of building from source was to obtain a .dll file, as while trying to include the Windows x64 release in my current project I kept getting a couple hundred linker errors, one example included below:

mavsdk.lib(curl_wrapper.obj) : error LNK2019: unresolved external symbol "__declspec(dllimport) public: class std::basic_istream<char,struct std::char_traits<char> > 

I have verified that I added the include directory and linked the .lib file appropriately for the environment I am working in, and got stuck there. The __declspec(dllimport) led me to believe that recompiling to add a .dll to the project might fix it, so now I am here. 

Finally, below is the entire output from start to finish of cloning, and then attempting to configure the repo in order to build it. 

==============================================================================================

Microsoft Windows [Version 10.0.19045.5608]
(c) Microsoft Corporation. All rights reserved.

C:\Users\logst>git clone https://github.com/mavlink/MAVSDK.git
Cloning into 'MAVSDK'...
remote: Enumerating objects: 58466, done.
remote: Counting objects: 100% (1605/1605), done.
remote: Compressing objects: 100% (564/564), done.
remote: Total 58466 (delta 1337), reused 1041 (delta 1041), pack-reused 56861 (from 3)
Receiving objects: 100% (58466/58466), 39.35 MiB | 5.94 MiB/s, done.
Resolving deltas: 100% (36988/36988), done.

C:\Users\logst>cd MAVSDK

C:\Users\logst\MAVSDK>git submodule update --init --recursive
Submodule 'mavsdk-proto' (https://github.com/mavlink/MAVSDK-Proto.git) registered for path 'proto'
Cloning into 'C:/Users/logst/MAVSDK/proto'...
Submodule path 'proto': checked out '81ffb7d2247354c5df9c57a42473652d303d576f'

C:\Users\logst\MAVSDK>cmake -DCMAKE_BUILD_TYPE=Release -Bbuild/default -S.
-- Building for: Ninja
-- Version: v3.2.0
-- Version major: 3
-- Version minor: 2
-- Version patch: 0
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/mingw64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning:
  Ignoring empty string ("") provided on the command line.
[... 715 lines cut ...]

CMake Warning (dev) at liblzma-stamp/download-liblzma.cmake:163:
  Syntax Warning in cmake code at column 22

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at liblzma-stamp/download-liblzma.cmake:167:
  Parse error.  Function missing ending ")".  Instead found unterminated
  string with text "

  )

  ".

CMake Error: Error processing file: C:/Users/logst/MAVSDK/build/default/third_party/liblzma/liblzma-prefix/src/liblzma-stamp/download-liblzma.cmake
ninja: build stopped: subcommand failed.
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  C:/Users/logst/MAVSDK/build/default/third_party/liblzma failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:14 (build_target)

-- Configuring incomplete, errors occurred!

C:\Users\logst\MAVSDK>

julianoes, 2025-03-27:
What cmake version is this?

In theory it should compile because it does so in CI. So if you replicate the steps and environment from GitHub Actions it "should just work".

By the way, you can just put ``` around code blocks.

julianoes, 2025-03-27:
Oh, I see, you're trying to build with MingW64. That's not officially supported. I'm sorry you're on your own. The recommendation is to use MSVC.

luterofromtx, 2025-04-01:
Thank you for your reply, I can't believe I missed that. I've since then went back and repeated the process using the developer command prompt for VS 2022 and went through the procedure of commands to build (both trying the Debug configuration, and Release configuration) as laid out at https://mavsdk.mavlink.io/main/en/cpp/guide/build.html using cmake 3.13.6. Unfortunately it still fails to build after successfully configuring.

For the first bit of output at the configuration stage I can see it's using MSVC now:
```
>cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -S.
-- Building for: Visual Studio 17 2022
-- Version: v3.2.0
-- Version major: 3
-- Version minor: 2
-- Version patch: 0
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.43.34809.0
-- The CXX compiler identification is MSVC 19.43.34809.0
```

For the linker warnings while building I'm getting many that look like 
``` 
LINK : warning LNK4286: symbol '?SetUpTestCase@Test@testing@@SAXXZ (public: static void __cdecl testing::Test::SetUpTes
tCase(void))' defined in 'gtest.lib(gtest-all.obj)' is imported by 'mavlink_component_metadata_test.obj'
```

And for linker errors I'm getting a ton of LNK2019 and some LNK2001 such as
```
mavlink_mission_transfer_client_test.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) class testi
ng::internal::FailureReporterInterface * __cdecl testing::internal::GetFailureReporter(void)" (__imp_?GetFailureReporte
r@internal@testing@@YAPEAVFailureReporterInterface@12@XZ) referenced in function "void __cdecl testing::internal::Asser
t(bool,char const *,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &
)" (?Assert@internal@testing@@YAX_NPEBDHAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
```
and
```
mavlink_mission_transfer_server_test.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) class testi
ng::internal::FailureReporterInterface * __cdecl testing::internal::GetFailureReporter(void)" (__imp_?GetFailureReporte
r@internal@testing@@YAPEAVFailureReporterInterface@12@XZ)
```
and ending with the last error message of 
```
fatal error LNK1120: 47 unresolved externals
```

I tried redoing the entire process over again start to finish but with the Release configuration, and I get a list of the same errors above during the build stage. These LNK2019 and LNK2001 errors also happen to be the same errors I was getting when originally trying to use the prebuilt winx64 release version in my project, where the errors came from the .lib file even adding the include directory and .lib file to the build tool in my project's environment. I'm not sure what else I could be doing wrong here.

julianoes, 2025-04-01:
I remember battling these. It's frustrating and I don't understand this stuff. Maybe that's why I don't develop for Windows.

Given the errors are with gtest, you can try building without testing:
```
-DBUILD_TESTING=OFF
```

https://github.com/mavlink/MAVSDK/blob/ddffc1e494723325e178d3fe8f189eb8b73aa266/third_party/CMakeLists.txt#L46-L48

luterofromtx, 2025-04-03:
Yeah, with the little experience I actually have I've already noticed that Windows seems to be its own headache to figure out with everything I've done so far. Thankfully building without testing worked in my case, so all seems to be good now. Thank you so much!!

closed 2025-05-06 by julianoes