mavlink/MAVSDK #2543: Uploading .plan to ArduPilot using import_qgroundcontrol_mission corrupts parameter

matthieu-db, 2025-04-16:
For the DIGICAM_CONTROL message, when uploading the .plan (attached as a .json) parameter 5 comes back when downloaded as 1.28e-5 instead of 1. The drone also doesn't take a picture (which it should because I am setting it to 1)

What could cause this? Is there a workaround?

[before upload to drone: testpretty.json](https://github.com/user-attachments/files/19778691/testpretty.json)

[after being downloaded from the drone: testdownloaded.json](https://github.com/user-attachments/files/19778742/testdownloaded.json)

Script I used to upload to the drone:

```
import asyncio

from mavsdk import System
import mavsdk.mission_raw

async def run():
    drone = System()
    await drone.connect(system_address="udp://0.0.0.0:14555")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    out = await drone.mission_raw.import_qgroundcontrol_mission(
        "testpretty.plan")

    print(f"{len(out.mission_items)} mission items and "
          f"{len(out.rally_items)} rally items imported.")

    await drone.mission_raw.upload_mission(out.mission_items)

    print("Mission uploaded")

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
```

matthieu-db, 2025-04-16:
I noticed this when testing our software (using MAVSDK-Java) with a real drone running ArduPilot 4.4.4. Then reproduced it with this minimal example using the Mission Planner SITL. and the above python script.

julianoes, 2025-04-29:
Not optimal, thanks for the issue.

Any chance you can make a unit test that shows it and fails?
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw/mission_import_test.cpp

julianoes, 2025-05-06:
Fix coming: #2560.

closed 2025-05-06 by julianoes

matthieu-db, 2025-05-15:
Thanks for your work, looking forward to testing it. Would have loved to make a unit test but I was on vacation until now

PR #2560 (2025-05-06): mission_raw: fix mission import for mission commands
We should only convert lat/lon values but not regular commands. We can tell by checking the mission item's frame.

Fixes #2543.