mavlink/MAVSDK #2549: Build failed with CMake 4.0

h4tr3d, 2025-04-18:
Error during confiugration:
```sh
cmake -S . -B buld
```

with result:
```sh
[100%] Built target mavlink
CMake Warning:
  Ignoring empty string ("") provided on the command line.

CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
CMake Error at third_party/cmake/build_target.cmake:32 (message):
  
  /mnt/zyncdev/user/NovaLabs/Hammer/repos/mavlink/mavlink-tools/mavsdk/third_party/jsoncpp
  failed to configure!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:12 (build_target)

-- Configuring incomplete, errors occurred!
```

Sounds like `third_party/jsoncpp/CMakeLists.txt` just should be updated to point to the cmake 3.5 at least. But I suggest change it to the same minimum cmake version, that used for root CMakeLists.txt for now: 3.13.

Same suggestion for all other third_party components, that point to the cmake 3.1.

Also, jsoncpp must be updated itself. It provides `src/jsoncpp/jsoncppConfig.cmake.in` that points into CMake 3.0 policy. Newest version updates it: https://github.com/open-source-parsers/jsoncpp/blob/master/jsoncppConfig.cmake.in

Another solution, just set ENV variable `CMAKE_POLICY_VERSION_MINIMUM` at the top-level CMakeLists.txt:
```cmake
set(ENV{CMAKE_POLICY_VERSION_MINIMUM} 3.13)
```
or run configuration like:
```sh
env CMAKE_POLICY_VERSION_MINIMUM=3.16 cmake -S . -B buld
```

dakejahl, 2025-04-18:
same as https://github.com/mavlink/MAVSDK/pull/2531

something to consider though is not forcing a cmake upgrade on everyone, especially since ubuntu 22.04 ships with 3.22.1

julianoes, 2025-05-06:
Should be fixed with https://github.com/mavlink/MAVSDK/pull/2542/commits/ea31f2d89bbb38516dd01c73452e17df0a57c9ae.

closed 2025-05-06 by julianoes