mavlink/MAVSDK #2553: Timeout error while uploading and downloading mission

NabinBhandari1, 2025-04-24:
https://github.com/mavlink/MAVSDK/issues/1316

    Hello @julianoes I am facing the same problem while uploading and downloading the mission file. Currently I'm using raspberry pi5. for running the script. my script work fine while running in laptop using UDP connection. But when I run the came script in pi5 changing the connection type to serial dev/ttyAMA0 i receive time out error. I also tried to build the custom mavsdk server as mentioned by you by changing "static constexpr double timeout_s = 2" still I reseive the time out error. I think I did some mistake while building the server. Could help me build the server for pi os with Raspberrry pi =5 architecture ?

julianoes, 2025-04-24:
Is the serial port on the RPi actually running? Can you confirm mavlink bytes are arriving?

NabinBhandari1, 2025-04-24:
I still get time out error while uploading mission, and terminal of server show, sending again after 0.507331s. I have attached the terminal output of server and code[image]

import json
import asyncio
from mavsdk import System
from mavsdk.mission import MissionItem, MissionPlan, MissionError

async def upload_mission_from_file(file_path):
    # Parse JSON mission file
    with open(file_path, 'r') as f:
        mission_data = json.load(f)

    # Extract mission items
    mission_items = []
    for item in mission_data['mission']['items']:
        if item['type'] == "SimpleItem":
            latitude = item['params'][4]
            longitude = item['params'][5]
            altitude = item['params'][6]
            is_fly_through = item['command'] == 16
            speed = mission_data['mission']['cruiseSpeed']
            mission_item = MissionItem(
                latitude_deg=latitude, longitude_deg=longitude, relative_altitude_m=altitude,
                speed_m_s=speed, is_fly_through=is_fly_through, gimbal_pitch_deg=0.0, gimbal_yaw_deg=0.0,
                camera_action=MissionItem.CameraAction.NONE, loiter_time_s=0, camera_photo_interval_s=0,
                acceptance_radius_m=5.0, yaw_deg=0.0, camera_photo_distance_m=0.0,
                vehicle_action=MissionItem.VehicleAction.NONE
            )
            mission_items.append(mission_item)
        elif item['type'] == "ComplexItem" and item['complexItemType'] == "fwLandingPattern":
            approach_lat, approach_lon, approach_alt = item['landingApproachCoordinate']
            land_lat, land_lon, land_alt = item['landCoordinate']
            approach_item = MissionItem(
                latitude_deg=approach_lat, longitude_deg=approach_lon, relative_altitude_m=approach_alt,
                speed_m_s=5.0, is_fly_through=False, gimbal_pitch_deg=0.0, gimbal_yaw_deg=0.0,
                camera_action=MissionItem.CameraAction.NONE, loiter_time_s=0, camera_photo_interval_s=0,
                acceptance_radius_m=5.0, yaw_deg=0.0, camera_photo_distance_m=0.0,
                vehicle_action=MissionItem.VehicleAction.NONE
            )
            mission_items.append(approach_item)
            land_item = MissionItem(
                latitude_deg=land_lat, longitude_deg=land_lon, relative_altitude_m=land_alt,
                speed_m_s=2.0, is_fly_through=False, gimbal_pitch_deg=0.0, gimbal_yaw_deg=0.0,
                camera_action=MissionItem.CameraAction.NONE, loiter_time_s=0, camera_photo_interval_s=0,
                acceptance_radius_m=5.0, yaw_deg=0.0, camera_photo_distance_m=0.0,
                vehicle_action=MissionItem.VehicleAction.LAND
            )
            mission_items.append(land_item)

    print(f"Total mission items: {len(mission_items)}")

    # Initialize drone without timeout
    drone = System()
    #await drone.connect(system_address="serial:///dev/ttyAMA0:38400")
    drone = System(mavsdk_server_address="localhost", port=50051)
    await drone.connect()
    
    
    # Wait for connection
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Drone connected!")
            break

    # Upload mission with retry and timeout
    max_attempts = 3
    for attempt in range(max_attempts):
        try:
            mission_plan = MissionPlan(mission_items)
            await asyncio.wait_for(drone.mission.upload_mission(mission_plan), timeout=60)
            print("Mission uploaded successfully!")
            await drone.mission.start_mission()
            break
        except asyncio.TimeoutError:
            if attempt < max_attempts - 1:
                print(f"Timeout occurred, retrying ({attempt + 1}/{max_attempts})...")
                await asyncio.sleep(2)
            else:
                print("Max attempts reached, mission upload failed.")
                raise
        except MissionError as e:
            print(f"Mission error: {e}")
            raise

# Run the mission upload
asyncio.run(upload_mission_from_file("updated_mission.plan"))

NabinBhandari1, 2025-04-29:
Issue has been solved.

julianoes, 2025-04-29:
Ok, what was the issue?

NabinBhandari1, 2025-04-30:
I tried changing double timeouts value but that too did not work,
Initially I was using AMA0 port (telem 2 to Rpi pin 6 8 and 10 ) for connecting Rpi and pixhawk, which was communicating slowly. After I switched to USB to USB connection using ACM0 and ACM1, the timeout problem was solved. Now there is no timeout issue now.

But using a USB connection for the Rpi and Pixhawk, I cannot use pixhawk's another usb port for connecting to GCS(QGC) on laptop  using USB cable for HITL test.

julianoes, 2025-04-30:
You will have to fix the RPi's AMA0 port. I suspect it is misconfigured and has the console or bluetooth or something conflicting. I definitely have had that working with RPi5 at a baudrate of 3000000

closed 2025-05-06 by julianoes