mavlink/MAVSDK #2554: MAVSDK/src/core unavailable while cloning MAVSDK

NabinBhandari1, 2025-04-25:
While cloning MAVSDK from https://github.com/mavlink/MAVSDK.git, the core and others submodule inside MAVSDK/src were not cloned. 

Even when I tried downloading the ZIP file, I was unable to get all the submodules. 

I also tried git clone --recurse-submodules https://github.com/mavlink/MAVSDK.git, but that did not work either.

JonasVautherin, 2025-04-25:
Did you try this?

```
git clone https://github.com/mavlink/mavsdk --recursive
```

NabinBhandari1, 2025-04-25:
Yes I tried , that as well but didn't work

JonasVautherin, 2025-04-25:
>  the core and others submodule inside MAVSDK/src were not cloned.

Wait, which submodules? The only submodule is `proto/`, at the root of the repo 🤔.

NabinBhandari1, 2025-04-26:
Actually, I'm encountering a timeout error when uploading or downloading missions.
I've reported the issue here: https://github.com/mavlink/MAVSDK/issues/2553.

I wanted to modify the double timeout = 0.5 value inside the mission_transfer.h file, but unfortunately, I couldn't find that file under src/core as mentioned in the MAVSDK GitHub repository. Instead, I found files like mission_transfer_server.h, mission_transfer_client.h, and mission_transfer_client_.h. I tried rebuilding the server after exploring those, but it didn’t resolve the issue.

Could you please guide me in resolving the timeout error?

This issue is inconsistent—sometimes the mission uploads successfully, and other times I get a timeout error, even when nothing changes (same code, same hardware, same connection type). I've shared the error output and mission upload script in the GitHub issue linked above.

julianoes, 2025-04-27:
To change the timeout, you should be able to change it through the API right here:
https://github.com/mavlink/MAVSDK/blob/7015261550bcb0cf14927bf373c487133e962c0a/src/mavsdk/core/include/mavsdk/mavsdk.h#L310-L318

No git cloning and compilation required, unless we miss something somewhere.

NabinBhandari1, 2025-04-27:
@julianoes Since I am working in MAVSDK using python, Its not working for me. Could you please guide me how to do this for python.

julianoes, 2025-04-29:
Oh, I see. We should expose this in Python then...

NabinBhandari1, 2025-04-29:
How to expose to python

NabinBhandari1, 2025-04-29:
Issue has been solved.

closed 2025-05-06 by julianoes