mavlink/MAVSDK #2555: when BUILD_WITHOUT_CURL=ON and BUILD_MAVSDK_SERVER=OFF

yygbruce, 2025-04-28:
After compile the mavsdk library with  BUILD_WITHOUT_CURL=ON and BUILD_MAVSDK_SERVER=OFF, and I build the offboard example, the following error occurs:
libmavsdk.so: undefined reference to `mavsdk::HttpLoader::~HttpLoader()'
libmavsdk.so: undefined reference to `mavsdk::HttpLoader::HttpLoader()'
libmavsdk.so: undefined reference to `mavsdk::HttpLoader::download_async(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (int, mavsdk::HttpStatus, CURLcode)>)'

julianoes, 2025-04-29:
Are you sure you are building against the correct library and header files? Or could it be a mismatch with libraries and headers installed on the system vs. locally?

yygbruce, 2025-04-29:
> Are you sure you are building against the correct library and header files? Or could it be a mismatch with libraries and headers installed on the system vs. locally?

yes, I build the library with default, it's ok.After, I set BUILD_WITHOUT_CURL=ON, then the error occurs.

julianoes, 2025-04-29:
Can you write down the cmake configure and build lines that you use, so I can exactly reproduce it?

julianoes, 2025-04-29:
And just make sure you don't have anything mavsdk left in `/usr/include` or `/usr/local/include`.

julianoes, 2025-04-29:
It looks like this was broken on `main`:

```
/home/julianoes/src/upstream/MAVSDK/src/mavsdk/plugins/camera/camera_impl.cpp: In member function ‘void mavsdk::CameraImpl::check_camera_definition_with_lock(PotentialCamera&)’:
/home/julianoes/src/upstream/MAVSDK/src/mavsdk/plugins/camera/camera_impl.cpp:1310:13: error: ‘notify_camera_list’ was not declared in this scope
 1310 |             notify_camera_list();
```

Once I fixed that I could reproduce your issue:

```
/usr/bin/ld: src/mavsdk/libmavsdk.so.3.2.0: undefined reference to `mavsdk::HttpLoader::~HttpLoader()'
```

PR #2556 (2025-05-01): Fix build without curl
Fixes #2555.
closed 2025-05-01 by julianoes