mavlink/MAVSDK #2561: I cannot access the 'Info' in the status text.

khosrooo, 2025-05-06:
### Environment:

    MAVSDK version: v3.2.0,v3.2.0

    Connection type: Serial (serial:///dev/ttyUSB0:115200)

    Operating System: Linux (Ubuntu 20.04)(jetson orin nx)

    Hardware: Ardupilote

my code is :
```
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <iostream>
#include <thread>
#include <chrono>

using namespace mavsdk;
using std::chrono::seconds;
using std::this_thread::sleep_for;

void usage(const std::string& bin_name)
{
    std::cerr << "Usage: " << bin_name << " <connection_url>\n"
              << "Connection URL format should be:\n"
              << " For TCP server: tcpin://<our_ip>:<port>\n"
              << " For TCP client: tcpout://<remote_ip>:<port>\n"
              << " For UDP server: udp://<our_ip>:<port>\n"
              << " For UDP client: udp://<remote_ip>:<port>\n"
              << " For Serial: serial://</path/to/serial/dev>:<baudrate>]\n"
              << "For example, to connect to the simulator use URL: udpin://0.0.0.0:14540\n";
}

int main(int argc, char** argv)
{
    if (argc != 2) {
        usage(argv[0]);
        return 1;
    }

    Mavsdk mavsdk{Mavsdk::Configuration{ComponentType::CompanionComputer}};
    ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }

    auto system = mavsdk.first_autopilot(3.0);
    if (!system) {
        std::cerr << "Timed out waiting for system\n";
        return 1;
    }

    auto telemetry = Telemetry{system.value()};

    // Subscribe to status text updates and print the text and severity
    telemetry.subscribe_status_text([](Telemetry::StatusText status_text) {
    std::cout << "Received status text update: " << status_text.text << '\n';
    std::string severity;
    switch (status_text.type) {
        case Telemetry::StatusTextType::Emergency: severity = "Emergency"; break;
        case Telemetry::StatusTextType::Alert: severity = "Alert"; break;
        case Telemetry::StatusTextType::Critical: severity = "Critical"; break;
        case Telemetry::StatusTextType::Error: severity = "Error"; break;
        case Telemetry::StatusTextType::Warning: severity = "Warning"; break;
        case Telemetry::StatusTextType::Notice: severity = "Notice"; break;
        case Telemetry::StatusTextType::Info: severity = "Info"; break;
        case Telemetry::StatusTextType::Debug: severity = "Debug"; break;
        default: severity = "Unknown"; break;
    }
    std::cout << "Status: [" << severity << "] " << status_text.text << '\n';
});

    // Keep the program running to continue receiving status text
    while (true) {
        sleep_for(seconds(1));
    }

    return 0;
}
```
### my output is 
07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:57|Debug] MAVLink: critical: EKF variance (system_impl.cpp:243)
Received status text update: EKF variance
Status: [Critical] EKF variance
[07:00:57|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:57|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:57|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:57|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:57|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)

"In the output, I see that 'Info' cannot be considered as a StatusText type, but 'Critical' can. Why is this the case? How can I access information of the 'Info' type? I think it is a bug. Although 'Info' is printed in the output, I cannot access it."

julianoes, 2025-05-06:
Hm, I'm trying to reproduce this but I don't see the problem:

```
cmake --build build && build/debug_statustext/debug_statustext udpin://0.0.0.0:14550
...
[09:42:48|Info ] MAVSDK version: v3.3.0 (mavsdk_impl.cpp:30)
[09:42:49|Info ] New system on: 127.0.0.1:39001 (with system ID: 1) (udp_connection.cpp:241)
[09:42:49|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:734)
[09:42:49|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[09:42:49|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:217)
[09:42:49|Debug] Discovered 1 component(s) (system_impl.cpp:497)
[09:42:53|Info ] heartbeats timed out (system_impl.cpp:265)
[09:42:55|Info ] New system on: 127.0.0.1:56849 (with system ID: 1) (udp_connection.cpp:241)
[09:42:55|Debug] MAVLink: debug: RCInput: decoding UDP (Pulses) (system_impl.cpp:243)
Received status text update: RCInput: decoding UDP (Pulses)
Status: [Debug] RCInput: decoding UDP (Pulses)
[09:42:55|Debug] MAVLink: info: Calibrating barometer (system_impl.cpp:243)
Received status text update: Calibrating barometer
Status: [Info] Calibrating barometer
[09:42:56|Debug] Discovered 1 component(s) (system_impl.cpp:497)
[09:42:56|Debug] MAVLink: info: ArduCopter V4.7.0-dev (fccd9600) (system_impl.cpp:243)
Received status text update: ArduCopter V4.7.0-dev (fccd9600)
Status: [Info] ArduCopter V4.7.0-dev (fccd9600)
[09:42:56|Debug] MAVLink: info: 8680c9f8f62e4276aaacc3bd02d098b6 (system_impl.cpp:243)
Received status text update: 8680c9f8f62e4276aaacc3bd02d098b6
Status: [Info] 8680c9f8f62e4276aaacc3bd02d098b6
[09:42:56|Debug] MAVLink: info: Frame: QUAD/PLUS (system_impl.cpp:243)
Received status text update: Frame: QUAD/PLUS
Status: [Info] Frame: QUAD/PLUS
[09:42:57|Debug] MAVLink: info: Barometer 1 calibration complete (system_impl.cpp:243)
Received status text update: Barometer 1 calibration complete
Status: [Info] Barometer 1 calibration complete
[09:42:57|Debug] MAVLink: info: Barometer 2 calibration complete (system_impl.cpp:243)
Received status text update: Barometer 2 calibration complete
Status: [Info] Barometer 2 calibration complete
[09:42:57|Debug] MAVLink: info: ArduPilot Ready (system_impl.cpp:243)
Received status text update: ArduPilot Ready
Status: [Info] ArduPilot Ready
[09:42:57|Debug] MAVLink: info: AHRS: DCM active (system_impl.cpp:243)
Received status text update: AHRS: DCM active
Status: [Info] AHRS: DCM active
[09:42:57|Debug] MAVLink: info: RC7: SaveWaypoint LOW (system_impl.cpp:243)
Received status text update: RC7: SaveWaypoint LOW
Status: [Info] RC7: SaveWaypoint LOW
[09:42:59|Debug] MAVLink: info: EKF3 IMU0 initialised (system_impl.cpp:243)
Received status text update: EKF3 IMU0 initialised
Status: [Info] EKF3 IMU0 initialised
[09:42:59|Debug] MAVLink: info: EKF3 IMU1 initialised (system_impl.cpp:243)
Received status text update: EKF3 IMU1 initialised
Status: [Info] EKF3 IMU1 initialised
[09:42:59|Debug] MAVLink: info: AHRS: EKF3 active (system_impl.cpp:243)
Received status text update: AHRS: EKF3 active
Status: [Info] AHRS: EKF3 active
[09:43:01|Debug] MAVLink: info: EKF3 IMU1 tilt alignment complete (system_impl.cpp:243)
Received status text update: EKF3 IMU1 tilt alignment complete
Status: [Info] EKF3 IMU1 tilt alignment complete
[09:43:01|Debug] MAVLink: info: EKF3 IMU0 tilt alignment complete (system_impl.cpp:243)
Received status text update: EKF3 IMU0 tilt alignment complete
Status: [Info] EKF3 IMU0 tilt alignment complete
[09:43:01|Debug] MAVLink: info: EKF3 IMU1 MAG0 initial yaw alignment complete (system_impl.cpp:243)
Received status text update: EKF3 IMU1 MAG0 initial yaw alignment complete
Status: [Info] EKF3 IMU1 MAG0 initial yaw alignment complete
[09:43:01|Debug] MAVLink: info: EKF3 IMU0 MAG0 initial yaw alignment complete (system_impl.cpp:243)
Received status text update: EKF3 IMU0 MAG0 initial yaw alignment complete
Status: [Info] EKF3 IMU0 MAG0 initial yaw alignment complete
[09:43:01|Info ] Removing inactive remote: 127.0.0.1:39001 (udp_connection.cpp:150)
[09:43:05|Debug] MAVLink: info: GPS 1: probing for u-blox at 230400 baud (system_impl.cpp:243)
Received status text update: GPS 1: probing for u-blox at 230400 baud
Status: [Info] GPS 1: probing for u-blox at 230400 baud
[09:43:05|Debug] MAVLink: info: GPS 1: detected u-blox (system_impl.cpp:243)
Received status text update: GPS 1: detected u-blox
Status: [Info] GPS 1: detected u-blox

```

I did find a segfault though with v3.3.0 that I'm fixing in #2563.

You might want to run your example with valgrind or ASAN to check for anything odd going on.

khosrooo, 2025-05-07:
thank u for responding 
I am using MAVSDK v3.3.0 to receive STATUSTEXT messages from an ArduPilot-based autopilot (running on a Jetson Orin NX with Ubuntu 20.04) over a serial connection (serial:///dev/ttyUSB0:115200). Intrinsic ArduPilot messages, such as EKF3 IMU0 initialised or RC10: EKFPosSource HIGH, are correctly captured by the subscribe_status_text callback in the Telemetry plugin. However, custom STATUSTEXT messages sent from a ground station GUI (e.g., "Grab.") are logged by MAVSDK but do not trigger the callback.

Status: [Info] EKF3 IMU0 initialised
is the instrinsic message from ardupilote but in my case  
07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:55|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[07:00:56|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)

i send custom message from ground station gui with this snip of code
```
 for i in range(0,5):
        try:
            # Ensure msg is a string and encode it
            if isinstance(msg, bytes):
                msg = msg.decode('utf-8')

            # Encode the msg as bytes
            msg_bytes = msg.encode('utf-8')

            # Ensure the msg is not longer than 50 characters (MAVLink limitation)
            msg_bytes = msg_bytes[:50]

            msg = the_connection.message_factory.statustext_encode(
                severity,
                msg_bytes
            )
            the_connection.send_mavlink(msg)
            logger.debug(f"Sent STATUSTEXT message: {msg}")
```
my problem i can not the accsess of custom message  
[08:21:30|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[08:21:31|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[08:21:31|Debug] MAVLink: info: Grab.: (system_impl.cpp:243)
[08:21:31|Debug] MAVLink: info: RC10: EKFPosSource HIGH (system_impl.cpp:243)
Received status text update: RC10: EKFPosSource HIGH
Status: [Info] RC10: EKFPosSource HIGH

julianoes, 2025-05-07:
Oh, now I understand. I think that's expected because those statustext messages come from a different System (different MAVLink system ID), and so they don't make it to the telemetry plugin which is instantiated for one system. If you want to subscribe to the ground station also, you would have to create a system for the ground station. Unfortunately, that's not really supported in MAVSDK-Python but really only in C++. You could start another mavsdk_server and Python script for that system potentially.

julianoes, 2025-05-20:
Going to close this. Comment if you need it re-opened.

closed 2025-05-20 by julianoes