mavlink/MAVSDK #2575: Windows 11 dependencies not found

dirksavage88, 2025-05-15:
I have tried both building the mavsdk library from source and using the generated one (zip file with the build artifacts). There are numerous issues and this is not an exhaustive list:

**1. The windows default app python does not work with the build: it must be uninstalled and a newer one installed from python.org**

**2. When configuring the build (`cmake -Bbuild -S.`), there is some a missing dependency:**
- Perl: can be installed via an msi installer package

**3. When building the mavsdk libraries from source, I followed the CI containers method by using:**
```
cmake -G "Visual Studio 17 2022" $env:superbuild -DCMAKE_PREFIX_PATH="$PWD/build/release/third_party/install" -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=build/release/install -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=OFF -DWERROR=ON -Bbuild/release -S.
```

->Otherwise there are a lot of linker errors when using the [recommended cmake](https://mavsdk.mavlink.io/main/en/cpp/guide/build.html#release) command for windows

**4. When linking to the generated mavsdk libraries cmake can not find curl (and by extension tinyxml2, jsoncpp, grpc).** Basically the dependencies in MAVSDKConfig.cmake are not found because they aren't installed to where cmake can find them in the base MAVSDK directory.

**5. There is a warning for windows to build with a local install in the Building C++ apps guide of the mavsdk docs, but this should explicitly say that system wide install is not supported.**

If this is the user experience for trying to develop on windows, can we really say it is supported or should there be an "at your own risk" warning?

julianoes, 2025-05-16:
Thanks for the issue @dirksavage88. Let's see if we can improve a few things.

1. What's the default Windows Python version?
2. Right, perl is required for openssl. We should add it to the docs: https://github.com/mavlink/MAVSDK/blob/main/docs/en/cpp/guide/build.md#windows. Do you want to make a pull request?
3. The additional CMAKE_PREFIX_PATH should no longer be required with my changes from yesterday #2568.
4. This is odd. I'm no expert on Windows but in Linux and macOS this just seems to work. I don't understand why this is so difficult on Windows.
5. Supported is a stretch indeed. We should add some sort of note. Or did you want to take on Windows maintenance and make sure it's properly supported?

JonasVautherin, 2025-05-16:
Did you check what `"$PWD/build/release/third_party/install"` resolves to? There are different terminals on Windows, `$PWD` doesn't work everywhere in my experience.

dirksavage88, 2025-05-16:
@JonasVautherin good question. So I am in visual studio 2022 and the default shell is a developer powershell.

That path does look like it resolves to the correct absolute path of the mavsdk build/release/install folder

@julianoes I can definitely make PR's to the docs for the first two issues. For the dependencies that are not found, do you have time to schedule a call to walk me through what the issue could be?

dirksavage88, 2025-05-16:
@JonasVautherin 

[image]

dirksavage88, 2025-05-16:
So for example if I go into one of the examples folders and try and build, it does't find MAVSDConfig.cmake unless I manually set the CMAKE_PREFIX_PATH, at that point it fails to find curl, tinxyml, egrpc, etc. I also tried setting the CMAKE_PREFIX_PATH directly to the third_party/install folder but still facing issues with libraries. 

Something tells me that  cmake didn't install things on the system correctly, but I could be wrong. I am following the build commands in the mavsdk docks for [building C++ apps](https://mavsdk.mavlink.io/main/en/cpp/guide/toolchain.html#building-the-app)

JonasVautherin, 2025-05-16:
In build/release/third_party/install/lib, do you see the libs, like curl and tinyxml?

CMAKE_PREFIX_PATH should point to /absolute/path/to/third_party/install. Make sure that the path is absolute 👍.

dirksavage88, 2025-05-19:
Using mavsdk v3.3.3 for both building from source and using the generated mavsdk libraries
```
Microsoft Visual Studio Community 2022
Version 17.13.6
VisualStudio.17.Release/17.13.6+35931.197
Microsoft .NET Framework
Version 4.8.09032

Installed Version: Community

Visual C++ 2022   00482-90000-00000-AA840
Microsoft Visual C++ 2022

ASP.NET and Web Tools   17.13.124.35287
ASP.NET and Web Tools

Azure App Service Tools v3.0.0   17.13.124.35287
Azure App Service Tools v3.0.0

C# Tools   4.13.0-3.25167.3+73eff2b5de2ad38ec602c0a9e82f9125fb85992b
C# components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Cookiecutter   17.0.25037.1
Provides tools for finding, instantiating and customizing templates in cookiecutter format.

GitHub Copilot   17.13.444.19527
GitHub Copilot is an AI pair programmer that helps you write code faster and with less work.

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

NuGet Package Manager   6.13.2
NuGet Package Manager in Visual Studio. For more information about NuGet, visit https://docs.nuget.org/

Python - Profiling support   17.0.25037.1
Profiling support for Python projects.

Python with Pylance   17.0.25037.1
Provides IntelliSense, projects, templates, debugging, interactive windows, and other support for Python developers.

Test Adapter for Boost.Test   1.0
Enables Visual Studio's testing tools with unit tests written for Boost.Test.  The use terms and Third Party Notices are available in the extension installation directory.

Test Adapter for Google Test   1.0
Enables Visual Studio's testing tools with unit tests written for Google Test.  The use terms and Third Party Notices are available in the extension installation directory.

TypeScript Tools   17.0.31211.2001
TypeScript Tools for Microsoft Visual Studio

Visual Basic Tools   4.13.0-3.25167.3+73eff2b5de2ad38ec602c0a9e82f9125fb85992b
Visual Basic components used in the IDE. Depending on your project type and settings, a different version of the compiler may be used.

Visual Studio IntelliCode   2.2
AI-assisted development for Visual Studio.

-- The C compiler identification is MSVC 19.43.34810.0
-- The CXX compiler identification is MSVC 19.43.34810.0

```

julianoes, 2025-05-20:
Ok, I'm trying to reproduce the issues. So far, I tried to build in the same way CI does:

```
cmake -G"Visual Studio 17 2022" -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=install -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=OFF -DWERROR=ON -Bbuild/release -S.

cmake --build build/release -j30 --config RelWithDebInfo --target install
```

The gRPC build fails, so I needed this fix: https://github.com/mavlink/MAVSDK/pull/2579

This produces a static mavsdk_server_bin.exe binary that runs without any special dependencies.

---

Now the step is to build examples against the installed library.

```
cmake -G"Visual Studio 17 2022" -DCMAKE_BUILD_TYPE=RelW thDebInfo -DCMAKE_PREFIX_PATH=../../install -DCMAKE_INSTALL_PREFIX=install -Bbuild/release -S.
```

And now I run into the problem that you mention:
```
-- Selecting Windows SDK version 10.0.26100.0 to target Windows 6.2.9200.
-- The C compiler identification is MSVC 19.44.35207.1
-- The CXX compiler identification is MSVC 19.44.35207.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Strawberry/c/share/cmake-3.29/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  C:/Users/Julian Oes/src/MAVSDK/install/lib/cmake/MAVSDK/MAVSDKConfig.cmake:7 (find_dependency)
  arm_authorizer_server/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
```

Let's see...

julianoes, 2025-05-20:
Ok, so it looks like the path of the dependencies is not properly exported, so for now we have to add the CMAKE_PREFIX_PATH of the mavsdk.lib build as well:

```
cmake -G"Visual Studio 17 2022" -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_PREFIX_PATH="../../install;../../build/release/third_party/install" -DCMAKE_INSTALL_PREFIX=install -Bbuild/release -S.
```

This almost works, except LibLZMA can't be found, so that's next to figure out.

(I disabled the example requiring SDL2 as that's separate and probably doesn't matter for our use case.)

dirksavage88, 2025-05-20:
`
This almost works, except LibLZMA can't be found, so that's next to figure out.
`

Yes, this is the latest issue I wasn't able to find a solution to. Looks related to https://github.com/microsoft/vcpkg/issues/14679

dirksavage88, 2025-05-20:
@julianoes just to confirm, we are seeing the same error when building example c++ apps?

```
PS C:\Users\dirks\source\MAVSDK\examples\mavlink_forwarding> cmake -G"Visual Studio 17 2022" -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_PREFIX_PATH="../../install;../../build/release/third_party/install" -DCMAKE_INSTALL_PREFIX=install -Bbuild/release -S.
-- The C compiler identification is MSVC 19.44.35207.1
-- The CXX compiler identification is MSVC 19.44.35207.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenSSL: C:/Users/dirks/source/MAVSDK/build/release/third_party/install/lib/libcrypto.lib (found suitable version "3.4.0", minimum required is "3")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- Found ZLIB: C:/Users/dirks/source/MAVSDK/build/release/third_party/install/lib/zlibstatic.lib (found version "1.3.0")
-- Found c-ares: C:/Users/dirks/source/MAVSDK/build/release/third_party/install/lib/cmake/c-ares/c-ares-config.cmake (found version "1.34.4")
-- Found RE2 via CMake.
-- Configuring done (5.0s)
CMake Error at C:/Users/dirks/source/MAVSDK/install/lib/cmake/MAVSDK/MAVSDKTargets.cmake:61 (set_target_properties):
  The link interface of target "MAVSDK::mavsdk" contains:

    LibLZMA::LibLZMA

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  C:/Users/dirks/source/MAVSDK/install/lib/cmake/MAVSDK/MAVSDKConfig.cmake:20 (include)
  CMakeLists.txt:12 (find_package)

-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

JonasVautherin, 2025-05-20:
Do you have liblzma somewhere in `C:/Users/dirks/source/MAVSDK/build/release/third_party/install`?

dirksavage88, 2025-05-20:
[image]

JonasVautherin, 2025-05-20:
That's something 👍.

Do you have something related to liblzma in `C:/Users/dirks/source/MAVSDK/build/release/third_party/install/lib/cmake`? Or maybe in `C:/Users/dirks/source/MAVSDK/build/release/third_party/install/lib/pkgconfig`?

dirksavage88, 2025-05-20:
[image]

I didn't find anything that matched "liblzma" in the pkgconfig folder (only zlib).

dirksavage88, 2025-05-20:
Also I noticed the liblzma version is 5.4.5? I am trying 5.8.1 just to see what happens. I have also seen an issue with Libevents but it was on another system (I moved to a faster laptop due to the long compile times)

julianoes, 2025-05-20:
> @julianoes just to confirm, we are seeing the same error when building example c++ apps?

Yes, exactly.

julianoes, 2025-05-20:
I'm currently battling cmake finding LibLZMA in my Strawberry Perl installation. How is Windows so messy using any path as a library location 🤯.

dirksavage88, 2025-05-20:
Got somewhere. Try this in MAVDK/install/lib/cmake/MAVSDK/MAVSDKConfig.cmake
```
include(CMakeFindDependencyMacro)

if(NOT OFF)
    if(${CMAKE_VERSION} VERSION_LESS "3.9.0")
        find_package(CURL REQUIRED CONFIG)
    else()
        find_dependency(CURL REQUIRED CONFIG)
    endif()

    find_dependency(Threads)
    find_dependency(jsoncpp)
    find_dependency(tinyxml2)
    find_dependency(LibLZMA)
    find_dependency(LibEvents)
    if(ON)
        find_dependency(gRPC)
    endif()
endif()

get_filename_component(MAVSDK_CMAKE_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
include("${MAVSDK_CMAKE_DIR}/MAVSDKTargets.cmake")

```

Then the command you used to configure the build for the example, then

`cmake --build build\release -j8 --config Release    `

 @julianoes:

julianoes, 2025-05-21:
I think that's probably what's missing indeed! Just double checking...

julianoes, 2025-05-21:
Yep, now that's looking better. Do you want to make a pull request or should I?

julianoes, 2025-05-21:
Here is my current state: https://github.com/mavlink/MAVSDK/pull/2581

This works for now, but I would also like to enable a build where you end up with a dll with everything statically built in, which would make this quite a bit easier.

dirksavage88, 2025-05-23:
Thanks @julianoes I will be testing it today

julianoes, 2025-05-24:
@dirksavage88 yes that makes sense.

With the changes coming in https://github.com/mavlink/MAVSDK/pull/2582, you should also be able to build with `-DBUILD_SHARED_LIBS=ON` and it should produce one DLL with everything included.

julianoes, 2025-06-02:
Going to close this as the mavsdk.dll generated by CI should now include everything and work out of the box. You find it in `bin/mavsdk.dll`.