mavlink/MAVSDK #2578: how to control the motor directly?

yygbruce, 2025-05-18:
Hi, @julianoes !I wan to control the motor directly, so i use the offboard::ActuatorControl, but i don't work.the code like the following:
mavsdk::Offboard::ActuatorControlGroup control_group_varied;
        control_group_varied.controls.resize(4);
        control_group_varied.controls[0] = 0.9f;
        control_group_varied.controls[1] = 0.9f;
        control_group_varied.controls[2] = 0.9f;
        control_group_varied.controls[3] = 0.9f;
        mavsdk::Offboard::ActuatorControl actuator_control_varied;
        actuator_control_varied.groups.push_back(control_group_varied);
        mavsdk::Offboard::Result send_actuator_varied_result = offboard_.set_actuator_control(actuator_control_varied);

julianoes, 2025-05-18:
Yes, unfortunately support for direct motor commands have been removed in PX4 at some point, I think around v1.14.

The recommendation is to use the ROS 2 bridge instead.

yygbruce, 2025-05-19:
ok!Thank you!

closed 2025-05-19 by yygbruce