mavlink/MAVSDK #2580: Issue of Mission Execution and Manual Input Priority

liu12344321, 2025-05-20:
When the "manual_control.set_manual_control_input"s value keeps changing, I want to execute a mission or other automode. In such a case, the value needs to remain consistent before and after the mission is triggered; otherwise, the mission will be interrupted. However, I’ve noticed that even if I set the manual value right before executing the mission, and then immediately trigger the task, it still gets interrupted. It turns out that I have to wait several hundred milliseconds after setting the value before executing the mission. It seems that the mission has a higher priority, even if the command is input later, the flight controller receives and processes it first.

Then I looked into the source code and found that these two types of commands follow different paths，one goes through a queue, while the other is sent directly as a command. So now I’m wondering: how can I make the information in the queue be delivered faster, so that I don’t have to wait for several hundred milliseconds?

Im current using v2.3.0
Thanks.

julianoes, 2025-05-21:
I'm not following. What are you trying to do and what is the problem exactly? And is this with PX4 or ArduPilot?

liu12344321, 2025-05-21:
@julianoes Im using PX4.
I want to implement the following use case: the drone is originally flying in mission mode, and my onboard vision computer monitors the environment. When an obstacle is detected, it will intervene and interrupt the mission. Once the obstacle is cleared and it's safe to proceed, I want to resume the mission.

The problem lies in resuming the mission. For example, initially I send set_manual_control_input(0, 0, 1, 0) to take control. Then, when I want to hand control back to the mission, I set set_manual_control_input(0, 0, 0.5, 0) and immediately call mission.start_mission(). After that, I continuously send set_manual_control_input(0, 0, 0.5, 0).

However, the resumed mission gets interrupted again right away. I believe this is because the earlier manual control input set_manual_control_input(0, 0, 1, 0) is still considered active by the flight controller. To avoid this, I have to wait a few hundred milliseconds after sending the first set_manual_control_input(0, 0, 0.5, 0) before calling mission.start_mission().

It seems like mission.start_mission() is being sent after but received before the manual control command by the flight controller.

julianoes, 2025-05-21:
Ok, I think manual_control is not the right interface to do this. Instead, I would pause the mission using `mission_pause` which will go into hold.

Or, alternatively if you want to do some control to avoid the obstacle, I would do that in offboard mode, e.g. using velocity commands.

Manual control is really meant for "manual" as in human control, and not automated control like that.

Also, PX4 implements an interface for obstacle avoidance that we probably should use instead, however, it's not implemented by MAVSDK yet, and I also don't know what the state of it is in PX4, but I do know that it would be the right thing to do, at least long term.

liu12344321, 2025-05-21:
@julianoes Because my drone is a VTOL operating in fixed-wing mode, it cannot hold position in place. So, when an obstacle is detected, I cannot simply switch to a HOLD mode. Instead, the onboard computer must automatically control the drone to fly toward a safe, obstacle-free area and resume mission. Although this is implemented via manual control commands as if a human were flying the drone, it is actually the onboard system issuing the commands autonomously.
I checked PX4's obstacle avoidance feature, and it currently only supports multicopter mode，so I want to implement an onboard computer to interrupt the mission when necessary and resume it when it's safe. But I have to add a delay of several hundred milliseconds  before resuming the mission, which is very troublesome.

julianoes, 2025-05-21:
Right. I'm afraid I don't have experience with all that.

I'm not 100% understanding your exact issue, but my understanding is that you would always send manual control commands, whether you are in mission or position mode, and then switch the mode between them as you see fit. Have you tried that?

julianoes, 2025-06-02:
Going to close this. If you have more comments or steps for me to reproduce this, just comment.

closed 2025-06-02 by julianoes