mavlink/MAVSDK #2584: How to use the lqr or mpc controller with mavsdk?

yygbruce, 2025-05-24:
Hi, @julianoes ! I want to use lqr or mpc controller with mavsdk to control the px4(gazebo simulation).The lqr controller is tested on ros1, it also be tested with mujoco(skydio_x2), but it don't work with mavsdk.Can you provide an example of how to customize the controller with mavsdk?

https://github.com/WillWu88/PX4-ROS-LQR

[px4ctrl_lqr.zip](https://github.com/user-attachments/files/20421778/px4ctrl_lqr.zip)

yygbruce, 2025-05-26:
> If you're using ROS, I suggest to use the lower level actuator interface that is provided by the [PX4 ROS 2 bridge](https://docs.px4.io/main/en/middleware/uxrce_dds.html).

Now, I want to use the mavsdk. The ros' version migration is not perfect.

julianoes, 2025-05-26:
In that case you need to use offboard control and there attitude, velocity, or position setpoints, as direct motor commands are no longer supported in PX4.