mavlink/MAVSDK #2588: MAVSDK 3.5.0 has issues with mavsdk-server and camera definition fetching

rayw-dronesense, 2025-05-27:
Tried dropping prebuilt .so and header files from [mavsdk_server_android-arm64.tar](https://github.com/mavlink/MAVSDK/releases/download/v3.5.0/mavsdk_server_android-arm64.tar) @ https://github.com/mavlink/MAVSDK/releases/tag/v3.5.0 into [mavsdk_server](https://github.com/mavlink/MAVSDK-Java/tree/main/mavsdk_server/src/main) and locally build, since we do not yet have a MAVSDK 3.x based Java SDK.

Ran into a couple of issues:

- mavsdk_server_run is not returning 'true', so `Java_io_mavsdk_mavsdkserver_MavsdkServer_runNative` is breaking early here
```
        if (!mavsdk_server_run(mavsdk_server, native_connection_url, mavsdk_server_port)) {
            return false;
        }
```
- bypassing this by ignoring the return value instead of returning false here, allows connecting to the drone; however, tinyxml seems to be erroring out when getting to camera definition,
```
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            E  Cannot open file "/data/data/com.dronesense.pilot.blue
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            D  Cache hit for camera_definition-ILX-LR1_Sony-65505.xml
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            E  Failed to open "/data/data/com.dronesense.pilot.blue
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            I  Using cached file "/data/data/com.dronesense.pilot.blue
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            E  tinyxml2::LoadFile failed: Error=XML_ERROR_FILE_READ_ERROR ErrorID=5 (0x5) Line number=0
2025-05-27 11:38:57.205 19034-19254 Mavsdk                  com.dronesense.pilot.blue            E  Failed to load camera definition: "/data/data/com.dronesense.pilot.blue
2025-05-27 11:38:57.209 19034-19546 Mavsdk                  com.dronesense.pilot.blue            W  Invalid positive gimbal ID: 1
```

Same config works correctly with MAVSDK 2.14.x / MAVSDK-Java 2.1.6

julianoes, 2025-05-27:
The API changed:

https://github.com/mavlink/MAVSDK/pull/2499

Is this what you're referring to?

matthieu-db, 2025-05-28:
@rayw-dronesense have you seen my PR that was merged yesterday regarding 3.5.0? https://github.com/mavlink/MAVSDK-Java/pull/202

rayw-dronesense, 2025-05-28:
@julianoes @matthieu-db I hadn't seen those before, thanks for letting me know.

However, the other issue should still be valid as the first issue was worked around by ignoring the return value and letting it continue.