mavlink/MAVSDK #2589: Mismatch between upload_mission_async() and subsequent download_mission_async()

cbelsito, 2025-05-29:
I am running Ardupilot SITL.
I am using `mavsdk::MissionRaw::upload_mission_async()` to upload a single `msvsdk::MissionRaw::MissionItem item`

```
item.seq = 1
item.frame = 3
item.command = 18   // MAV_CMD_NAV_LOITER_TURNS
item.current = 0
item.autocontinue = 1
item.param1 = 5
item.param2 = 1
item.param3 = 100
item.param4 = 0
item.x = 420000000
item.y = 100000000
item.z = 100
item.mission_type = 0
```

After that, I am calling `mavsdk::MissionRaw::download_mission_async()` to verify that the correct data was uploaded. Every value downloaded from Ardupilot SITL is the same except for `param2` where I get a mismatch:
```
item.seq = 1
item.frame = 3
item.command = 18   // MAV_CMD_NAV_LOITER_TURNS
item.current = 0
item.autocontinue = 1
item.param1 = 5
item.param2 = 0
item.param3 = 100
item.param4 = 0
item.x = 420000000
item.y = 100000000
item.z = 100
item.mission_type = 0
```

Interestingly enough, if I set `item.param4 = 1` and upload, then the downloaded MissionItem will also have `item.param4 == 1` so to me it seems an issue specifically with `item.param2`

cbelsito, 2025-05-29:
I also tried to change the Loiter-Type to `MAV_CMD_NAV_LOITER_TO_ALT`
UPLOAD:
```
item.seq = 1
item.frame = 3
item.command = 31   // MAV_CMD_NAV_LOITER_TO_ALT
item.current = 0
item.autocontinue = 1
item.param1 = 1
item.param2 = 100
item.param3 = 0
item.param4 = 1
item.x = 420000000
item.y = 100000000
item.z = 100
item.mission_type = 0
```

DOWNLOAD:
```
item.seq = 1
item.frame = 3
item.command = 31   // MAV_CMD_NAV_LOITER_TO_ALT
item.current = 0
item.autocontinue = 1
item.param1 = 0
item.param2 = 100
item.param3 = 0
item.param4 = 1
item.x = 420000000
item.y = 100000000
item.z = 100
item.mission_type = 0
```

For `MAV_CMD_NAV_LOITER_TO_ALT` the "Heading Required" property moves according to MAVLINK documentation from param2 to param1, the mismatch here occurs with param1. It seems that the "Heading Required" property has some issues?

julianoes, 2025-05-30:
Do you think this gets lost on the MAVSDK side or ArduPilot side? Any chance you can double check using QGC?

cbelsito, 2025-05-31:
Hi, I just verified with QGC and it seems it shows the same behaviour. I am sending a Loiter-Waypoint with "Exit once aligned" but if I restart QGC and verify the mission it actually always shows "Exit in any direction". I am not 100% but maybe it's in fact a Ardupilot SITL problem not handling certain waypoint param values correctly?

JonasVautherin, 2025-06-01:
Could it be an Ardupilot bug?

julianoes, 2025-06-01:
Ok, in this case I'll close this.

closed 2025-06-01 by julianoes