mavlink/MAVSDK #2590: Camera server freezes up and disconnects upon taking photo or record video 3.5

jnomikos, 2025-05-30:
Issue I am seeing:

Running MAVSDK 3.5.0's camera server example does not work for me as expected. The camera does show up, but after taking 1 or 2 photos or starting recording, photo/video controls freeze up and then the MAVLink camera disconnects. This was not happening on earlier versions.

I saw this both happen on ArduCopter 4.5.6 and mainline PX4 simulation. 

I also witnessed this happen on both QGC 4.4 and release candidate for 5.0.

https://github.com/user-attachments/assets/ccc70541-3a38-4da0-b6fb-83f6b3bdbf8c

The one major difference between my tests on PX4 and Ardupilot was that on ardupilot, occasionally the camera status would not be detected as idle, but this only sometimes happened. Ardupilot and PX4 both had the camera server freezing up issue. 

System Specs:

Ubuntu 22.04
MAVSDK 3.5
QGC 4.4 & Mainline
Mainline PX4 & Arducopter 4.5.6

jnomikos, 2025-05-30:
PX4 1.15.2 has same issue as well. In the video I did not notice it at first, but MAVLink camera controls do disappear here too.

https://github.com/user-attachments/assets/6d72f302-fcc4-4981-b0b4-b3978dd9fcf1

julianoes, 2025-05-30:
When it "freezes up", any chance you can attach gdb to it and get a backtrace of all threads?

So I would do:

1. `ps aux | grep camera_server`
2. `sudo gdb build/examples/camera_server`
3. Then type `attach <PID from first step>
4. `thread apply all bt`

I suspect it's some sort of lockup in the thread locks, or in the example code.

jnomikos, 2025-05-30:
@julianoes I'll try doing that.

Did further testing, the issue got introduced in the 3.3 release it seems. I do not see the issue in 3.2 but I see it in 3.3

jnomikos, 2025-05-30:
Here you go @julianoes 

This is on 3.5.0 again

```
(gdb) thread apply all bt

Thread 8 (Thread 0x7ebc15ffb640 (LWP 222570) "camera_server"):
#0  0x00007ebc1d8e57f8 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7ebc15ffacf0, rem=0x7ebc15ffacf0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007ebc1d8ea677 in __GI___nanosleep (req=<optimized out>, rem=<optimized out>) at ../sysdeps/unix/sysv/linux/nanosleep.c:25
#2  0x00005a722864aec2 in std::this_thread::sleep_for<long, std::ratio<1l, 1000l> > (__rtime=...) at /usr/include/c++/11/bits/this_thread_sleep.h:82
#3  0x00007ebc1ed89b93 in mavsdk::SystemImpl::system_thread (this=0x7ebc18003240) at /home/john/ssd2/work/repos/MAVSDK/src/mavsdk/core/system_impl.cpp:293
#4  0x00007ebc1edb2046 in std::__invoke_impl<void, void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> (__f=@0x7ebc18002e90: (void (mavsdk::SystemImpl::*)(mavsdk::SystemImpl * const)) 0x7ebc1ed899aa <mavsdk::SystemImpl::system_thread()>, __t=@0x7ebc18002e88: 0x7ebc18003240) at /usr/include/c++/11/bits/invoke.h:74
#5  0x00007ebc1edb1f99 in std::__invoke<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> (__fn=@0x7ebc18002e90: (void (mavsdk::SystemImpl::*)(mavsdk::SystemImpl * const)) 0x7ebc1ed899aa <mavsdk::SystemImpl::system_thread()>) at /usr/include/c++/11/bits/invoke.h:96
#6  0x00007ebc1edb1ef9 in std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> >::_M_invoke<0ul, 1ul> (this=0x7ebc18002e88) at /usr/include/c++/11/bits/std_thread.h:259
#7  0x00007ebc1edb1eae in std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> >::operator() (this=0x7ebc18002e88) at /usr/include/c++/11/bits/std_thread.h:266
#8  0x00007ebc1edb1e8e in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (mavsdk::SystemImpl::*)(), mavsdk::SystemImpl*> > >::_M_run (this=0x7ebc18002e80) at /usr/include/c++/11/bits/std_thread.h:211
#9  0x00007ebc1dcdc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ebc1d894ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#11 0x00007ebc1d926850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

Thread 7 (Thread 0x7ebc167fc640 (LWP 222569) "camera_server"):
#0  __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7ebc180038e0) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x7ebc180038e0) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x7ebc180038e0, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:139
#3  0x00007ebc1d893a41 in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7ebc18003890, cond=0x7ebc180038b8) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x7ebc180038b8, mutex=0x7ebc18003890) at ./nptl/pthread_cond_wait.c:627
#5  0x00007ebc1ef0d2f9 in mavsdk::SafeQueue<std::shared_ptr<mavsdk::HttpLoader::WorkItem> >::dequeue (this=0x7ebc18003840) at /home/john/ssd2/work/repos/MAVSDK/src/mavsdk/core/safe_queue.h:36
#6  0x00007ebc1ef0c532 in mavsdk::HttpLoader::work_thread (self=0x7ebc18003830) at /home/john/ssd2/work/repos/MAVSDK/src/mavsdk/core/http_loader.cpp:58
#7  0x00007ebc1ef111aa in std::__invoke_impl<void, void (*)(mavsdk::HttpLoader*), mavsdk::HttpLoader*> (__f=@0x7ebc18004690: 0x7ebc1ef0c4f6 <mavsdk::HttpLoader::work_thread(mavsdk::HttpLoader*)>) at /usr/include/c++/11/bits/invoke.h:61
#8  0x00007ebc1ef1111d in std::__invoke<void (*)(mavsdk::HttpLoader*), mavsdk::HttpLoader*> (__fn=@0x7ebc18004690: 0x7ebc1ef0c4f6 <mavsdk::HttpLoader::work_thread(mavsdk::HttpLoader*)>) at /usr/include/c++/11/bits/invoke.h:96
#9  0x00007ebc1ef1107d in std::thread::_Invoker<std::tuple<void (*)(mavsdk::HttpLoader*), mavsdk::HttpLoader*> >::_M_invoke<0ul, 1ul> (this=0x7ebc18004688) at /usr/include/c++/11/bits/std_thread.h:259
#10 0x00007ebc1ef10fba in std::thread::_Invoker<std::tuple<void (*)(mavsdk::HttpLoader*), mavsdk::HttpLoader*> >::operator() (this=0x7ebc18004688) at /usr/include/c++/11/bits/std_thread.h:266
#11 0x00007ebc1ef10f0a in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(mavsdk::HttpLoader*), mavsdk::HttpLoader*> > >::_M_run (this=0x7ebc18004680) at /usr/include/c++/11/bits/std_thread.h:211
#12 0x00007ebc1dcdc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ebc1d894ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#14 0x00007ebc1d926850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

Thread 6 (Thread 0x7ebc16ffd640 (LWP 222568) "camera_server"):
#0  0x00007ebc1d8e57f8 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7ebc16ffccf0, rem=0x7ebc16ffccf0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007ebc1d8ea677 in __GI___nanosleep (req=<optimized out>, rem=<optimized out>) at ../sysdeps/unix/sysv/linux/nanosleep.c:25
#2  0x00005a722864aec2 in std::this_thread::sleep_for<long, std::ratio<1l, 1000l> > (__rtime=...) at /usr/include/c++/11/bits/this_thread_sleep.h:82
#3  0x00007ebc1ed89b93 in mavsdk::SystemImpl::system_thread (this=0x7ebc18000ef0) at /home/john/ssd2/work/repos/MAVSDK/src/mavsdk/core/system_impl.cpp:293
[... cut ...]
#0  0x00007ebc1d8e57f8 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffe91d8000, rem=0x7fffe91d8000) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007ebc1d8ea677 in __GI___nanosleep (req=<optimized out>, rem=<optimized out>) at ../sysdeps/unix/sysv/linux/nanosleep.c:25
#2  0x00005a722864ac4d in std::this_thread::sleep_for<long, std::ratio<1l, 1l> > (__rtime=...) at /usr/include/c++/11/bits/this_thread_sleep.h:82
--Type <RET> for more, q to quit, c to continue without paging--
#3  0x00005a7228646cc9 in main (argc=1, argv=0x7fffe91d8308) at /home/john/ssd2/work/repos/MAVSDK/examples/camera_server/camera_server.cpp:58
```

julianoes, 2025-06-01:
Perfect, thanks!

So, it's blocked here:
https://github.com/mavlink/MAVSDK/blob/b1b219808572b328481179a183f7f7226079eab2/src/mavsdk/plugins/camera_server/camera_server_impl.cpp#L1314

And presumably here:
https://github.com/mavlink/MAVSDK/blob/b1b219808572b328481179a183f7f7226079eab2/src/mavsdk/plugins/camera_server/camera_server_impl.cpp#L1056

I need to look at it for a bit longer to figure it out... but definitely some sort of deadlock going on. Maybe thread sanitizer will illustrate it better.

jnomikos, 2025-06-01:
> Ok, trying to reproduce this. How did you connect it to PX4 exactly?

1. Run PX4 SITL `make px4_sitl gz_x500`
2. I switched the UDP port on the example to 14550 since 14030 wasn't working
3. Connect via QGC, try taking photo, record video

I also tested on Ardupilot by:

1. Connected to cube orange flight controller to computer via usb
2. Used MAVLink Router to take in serial from cube orange and both output to 14550 and default tcp port
3. Run camera server example with udp 14550, and connect to QGroundControl via TCP comm link
4. Witnessed same issue.

julianoes, 2025-06-01:
Thanks, just found that as well. The port was wrong, it should be `14280` in the camera_server.

julianoes, 2025-06-01:
For me it seems to get stuck earlier than that:

```
cmake --build build-tsan && build-tsan/camera_server/camera_server
ninja: no work to do.
[05:10:23|Info ] MAVSDK version: v3.5.0-4-gb1b21980 (mavsdk_impl.cpp:30)
Created camera server connection
[05:10:24|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:818)
[05:10:24|Debug] Component Autopilot (1) added. (system_impl.cpp:376)
[05:10:24|Debug] New system ID: 255 Comp ID: 190 (mavsdk_impl.cpp:818)
[05:10:24|Debug] Component Ground station (190) added. (system_impl.cpp:376)
[05:10:24|Debug] Discovered 1 component(s) (system_impl.cpp:494)
[05:10:24|Warn ] Got unknown request message! (camera_server_impl.cpp:1087)
[05:10:25|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:216)
[05:10:25|Debug] Discovered 1 component(s) (system_impl.cpp:494)
[05:10:27|Warn ] Got unknown request message! (camera_server_impl.cpp:1087)
[05:10:28|Warn ] Callback took more than 1 second to run.
See: https://mavsdk.mavlink.io/main/en/cpp/troubleshooting.html#user_callbacks (mavsdk_impl.cpp:956)
```

I can't even switch to photo mode.

jnomikos, 2025-06-01:
Interesting. It's possible when I was testing that the mode switch also timed out for me, but I saw it take the first photo because the GCS didn't detect the timeout until later. Not sure though.

> For me it seems to get stuck earlier than that:
>
> ```
> cmake --build build-tsan && build-tsan/camera_server/camera_server
> ninja: no work to do.
> [05:10:23|Info ] MAVSDK version: v3.5.0-4-gb1b21980 [...]

julianoes, 2025-06-01:
This fixes it for me: [#2591](https://github.com/mavlink/MAVSDK/pull/2591)

jnomikos, 2025-06-01:
Ah nice catch, I see where the double lock happened now. I'll test it out on my own end

closed 2025-06-01 by julianoes