mavlink/MAVSDK #2593: MAVSDK for ardupilot gimbal control issue after soft reboot

jnomikos, 2025-06-03:
From my experience, MAVSDK does not detect a system is lost on soft reboot. Which is fine for most things I am doing. camera_server, param, and mavlink_passthrough seem to work fine for me even after the soft reboot.

The problem is that gimbal does not work after the reboot. Giving any gimbal control after soft rebooting will result in a failure to do so, with it returning the result as "Error". 

My assumption is that MAVSDK still thinks it has control of the gimbal, when it doesn't, since it did not know the autopilot was rebooted.

I am not able to make it so that MAVSDK detects reboots, even if I increase BRD_BOOT_DELAY. 

In my application, this will be fine, because I have a background thread that detects when we last received telemetry position, so I can use that to detect a reboot, but I still wanted to make an issue here, just to give a heads up of the issue I found. 

System:

MAVSDK v3.2.0
ArduCopter 4.5.6
Pixy LR gimbal

Note: MAVSDK is connected to flight controller via UART

jnomikos, 2025-06-03:
Nevermind, the issue is intermittent and doesn't seem to be triggered by soft reboot.

Also, for some reason, it only happens to me when connection is serial. When it's udpout instead with mavlink-router, no issue.

jnomikos, 2025-06-03:
Sounds good @julianoes . Since I'm not confident on the source of this problem, I closed the issue. I'll reopen if I find out more info

jnomikos, 2025-06-04:
Looks like it continually requesting GIMBAL_MANAGER_INFORMATION from 1/154 was causing this problem. It seemed like that retries mechanism was not really working either and it would indefinitely request it. Now that I have that hardcoded away, things are not erroring. Hmmm

jnomikos, 2025-09-04:
Update: Not sure the reason why this happens on ArduPilot. Will keep this issue open just in case it happens with other gimbals. It could be an issue specific to Gremsy Pixy LR though, I am not completely sure.

julianoes, 2025-11-23:
No sure what to do here. Comment again if something needs doing.

closed 2025-11-23 by julianoes