mavlink/MAVSDK #2611: Double Unlock in LockedQueue::Guard

grantwilk, 2025-07-13:
## Problem

- The `LockedQueue::Guard` class locks the mutex in constructor and unlocks in destructor
- `wait_and_pop_front()` manually unlocks the mutex, then re-locks with `unique_lock`
- When the method returns, both the `unique_lock` and `Guard` get destroyed as they leave scope
- The `unique_lock` destructor AND `Guard` destructor try to unlock the same mutex
- This causes undefined behavior (double-unlock) that can crash/hang the program

Personally I experienced this causing indefinite hangs in my application.

## History

- First observed in v3.7.0 release
- Does not reproduce in v3.6.0 release
- Likely introduced in commit https://github.com/mavlink/MAVSDK/commit/1abe3ea074e3c340d6c26865788a5980826a1c8e by @julianoes

julianoes, 2025-07-13:
Oh dear, sorry about that. And thanks for the issue. Clearly, I didn't review this carefully enough.

grantwilk, 2025-07-13:
No worries, I appreciate the improvements toward thread safety! The 3.6.0 release works great until then.