mavlink/MAVSDK #2612: Accessing mock MAVSDK classes

mattforestyoung, 2025-07-14:
I'm using MAVSDK for a project, and I'd like to access the mock classes provided for `Telemetry`, `System` etc when writing some unit tests for it. From what I can tell these classes aren't installed along with the regular headers, so they can't be accessed through the normal installation process. And so I've been thinking about different ways to include them.

One option might be to add an additional compilation flag to the CMakeLists files, something like the code below: 

```
if (EXPORT_MOCK_HEADERS)
  install(FILES
      mocks/telemetry_mock.h
      DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/mavsdk/plugins/telemetry)
endif()
```

If I offered to add this feature, would it be something the MAVSDK developers want and see value in? Or is it unnecessary and you would recommend just including the mock headers directly from the source code?

julianoes, 2025-07-14:
My first hunch is that I don't want to expose it, but I'm not using mocks much, so I should have @JonasVautherin weigh in instead.

Also, if it's using a cmake option, that's not so bad.

JonasVautherin, 2025-07-14:
I would think that this is rather unusual 🤔. The way I would approach it in a project would be by wrapping MAVSDK into my own structure and mocking that.

Is there a disadvantage for you in just copying the mock headers you want? It feels like it's probably more practical because then you don't have to rebuild MAVSDK from sources just for copying them...

mattforestyoung, 2025-07-14:
> The way I would approach it in a project would be by wrapping MAVSDK into my own structure and mocking that.

That's what I was planning on doing, but then I saw that you've already gone through the effort of creating all those mock classes; it seemed like a shame to not reuse them and save myself some effort.

> Is there a disadvantage for you in just copying the mock headers you want? It feels like it's probably more practical because then you don't have to rebuild MAVSDK from sources just for copying them...

Not really, it just means I have to be careful about updating those headers if I choose to update the MAVSDK version I'm using, in case there is a new change not reflected in the copied mock headers.

I'm happy to run with that solution, I just wanted to make the suggestion in case it was something you saw value in adding.

JonasVautherin, 2025-07-15:
> I just wanted to make the suggestion in case it was something you saw value in adding.

Yeah thanks a lot for asking! The thing is that adding if-branches into the CMakeLists makes it harder to maintain, and it doesn't seem like reusing our mocks is a common thing 🙈.

mattforestyoung, 2025-07-15:
Totally get it, CMake can be a pain to deal with at the best of times, and you're right that most people probably don't have the time or need for mock testing.

closed 2025-07-15 by mattforestyoung