mavlink/MAVSDK #2615: add_any_connection not working as intended

ehenry344, 2025-07-16:
Currently trying to get some of the examples to run on my Windows 11 system and haven't been successful.

I'm building the examples with the local install of MAVSDK (from mavsdk-windows-x64-release) rather then building from source, and including the mavsdk.dll in the same directory as the executable for each example. When trying to run each example as intended ex: "./build/Debug/gimbal_device_tester.exe udpin://0.0.0.0:14540" Ill get an error saying that the protocol / connection URL is invalid, and when switching to "udpin://:14540" the error doesn't appear, but when `add_any_connection` is called during execution it hangs indefinitely, and eventually causes the program to abort.

Same behavior is observed running other examples such as: takeoff_and_land, autopilot_server, etc.

julianoes, 2025-07-17:
Have you tried putting some quotes or double quotes around the argument `udpin://0.0.0.0:14540`?

ehenry344, 2025-07-17:
Yes, I have done this aswell.

julianoes, 2025-07-17:
This is where it ends up: https://github.com/mavlink/MAVSDK/blob/v3.7.1/src/mavsdk/core/cli_arg.cpp#L62

ehenry344, 2025-07-17:
Yes, i've used both.

guo543, 2025-07-20:
Hi, I'm having the same issue on Windows, I have a very simple main function which creates the MAVSDK instance and add udp connections.
```cpp
int main(int argc, char** argv) {
  if (argc < 2) {
    return 1;
  }
  Mavsdk mavsdk{Mavsdk::Configuration{ComponentType::GroundStation}};
  ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);
  if (connection_result == mavsdk::ConnectionResult::Success) {
    std::cout << "success\n";
  }
  return 0;
}
```

After building I manually copied mavsdk.dll to the same directory as the executable.

In powershell when I run:
```
.\app.exe udpin://0.0.0.0:14540
```
I get
```
[12:15:51|Info ] MAVSDK version: v3.7.2 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:30)
[12:15:51|Error] Unknown protocol (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\cli_arg.cpp:62)
```
I get the same output when I run
```
.\app.exe udp://0.0.0.0:14540
```
Adding quotes also does not solve this.

Thank you in advance.

julianoes, 2025-07-21:
Could it be ABI incompatibility between Debug and Release builds?

https://stackoverflow.com/questions/73895291/why-cant-we-mix-debug-and-release-builds-with-msvc
https://stackoverflow.com/questions/73895291/why-cant-we-mix-debug-and-release-builds-with-msvc

guo543, 2025-07-22:
> Could it be ABI incompatibility between Debug and Release builds?
> 
> https://stackoverflow.com/questions/73895291/why-cant-we-mix-debug-and-release-builds-with-msvc https://claude.ai/share/ff5eb28b-0e66-401e-9ba5-4721cec511f3

Hi Julian,
Thank you for your response. I believe this is causing the problem.

julianoes, 2025-07-22:
Ok, I'm adding a note in the docs about this.

@ehenry344 please check if this fixes it for you.

ehenry344, 2025-07-22:
I was able to build and run the examples on my Mac without issue. Currently trying to build on my Windows machine via VS Dev CMD Prompt and cannot get past the openssl build / install step. Where the following error is produced:

```[3/4] Performing install step for 'openssl'
FAILED: openssl/src/openssl-stamp/openssl-install C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl/openssl/src/openssl-stamp/openssl-install
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\evan\Documents\MAV\MAVSDK\Build\third_party\openssl\openssl\src\openssl-build && nmake install_dev && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl/openssl/src/openssl-stamp/openssl-install"

Microsoft (R) Program Maintenance Utility Version 14.44.35211.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1065: invalid option '-'
Stop.
ninja: build stopped: subcommand failed.
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl
  failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:40 (build_target)```

Hoping to figure this out, and attempting to run the examples.

ehenry344, 2025-07-22:
As mentioned it seems the incompatibility between doing a debug build of the examples with the release binaries was the reason for this issue. Considering:

After running: ```cmake --build build --config Release -j4``` over ```cmake --build build -j4``` I was able to get the examples up and running using the release binaries.

A similar situation is documented in: https://github.com/mavlink/MAVSDK/issues/1902 and as @julianoes mentioned I think it would be worthwhile to include this detail regarding properly ensuring build config is set appropriately. 

Thank you both for your time given to the matter.

closed 2025-07-22 by ehenry344

guo543, 2025-07-22:
> I was able to build and run the examples on my Mac without issue. Currently trying to build on my Windows machine via VS Dev CMD Prompt and cannot get past the openssl build / install step. Where the following error is produced:
> 
> ```
> FAILED: openssl/src/openssl-stamp/openssl-install C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl/openssl/src/openssl-stamp/openssl-install
> C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\evan\Documents\MAV\MAVSDK\Build\third_party\openssl\openssl\src\openssl-build && nmake install_dev && "C:\Program Files\CMake\bin\cmake.exe" -E touch C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl/openssl/src/openssl-stamp/openssl-install"
> 
> Microsoft (R) Program Maintenance Utility Version 14.44.35211.0
> Copyright (C) Microsoft Corporation.  All rights reserved.
> 
> NMAKE : fatal error U1065: invalid option '-'
> Stop.
> ninja: build stopped: subcommand failed.
> CMake Error at third_party/cmake/build_target.cmake:51 (message):
>   C:/Users/evan/Documents/MAV/MAVSDK/Build/third_party/openssl
>   failed to build!
> Call Stack (most recent call first):
>   third_party/CMakeLists.txt:40 (build_target)```
> 
> Hoping to figure this out, and attempting to run the examples.
> ```

Hi,
I know this issue is closed and we should probably start a new one for this, but I was having exactly the same issue when building on Windows. Setting -DBUILD_WITHOUT_CURL=OFF for cmake will let it skip openssl. But I also hope there's a proper way to solve this. I heard about MAKEFLAGS [here](https://stackoverflow.com/questions/1975240/nmake-exe-keeps-complaining-about-flags-im-not-giving-it-u1065) but it is not set for me on Windows.

julianoes, 2025-07-22:
@guo543 please create a new separate issue and provide the steps for me to reproduce, thanks.