mavlink/MAVSDK #2620: use gimbal plugin's gibal_list and attitude function, the gimbal_id is not match.

jdx200512, 2025-07-23:
use gimbal plugin's gibal_list and attitude function, the gimbal_id is not match.
I try to control camera gimbal through mavsdk-server and mavsdk-python.
the system only has one gimbal, gimbal_list return gimbal_id is 1, but somtimes I can not control this gimbal use gimbal_id:1.
I get retsult from attitude the gimbal_id is 3, and I can control the gimbal with gimbal_id:3.

[image]

mavsdk_server version: 3.6.0

[image]

jdx200512, 2025-07-23:
> What sort of gimbal is this and how is it connected and set up?

gimbal sort: SIYI A8mini
uart telem2

[image]

MNT_MODE_IN and OUT both with MAVlink gimbal protocol v2
[image]

jdx200512, 2025-07-23:
> Can you get the same logs with env variable `MAVSDK_GIMBAL_DEBUGGING=1`? I'm not quite sure what is going on.
> 
> Also, what SIYI gimbal version is this?

the version of SIYI gimbal as follow

[image]

[image]

julianoes, 2025-07-23:
Thanks!

Can you get the full output of the gimbal list?

My guess is that there is one message sent by the gimbal device (SIYI) and one by PX4, but I'm not 100% sure.

I have this gimbal on my desk, so I will have to try to reproduce this tomorrow.

jdx200512, 2025-07-24:
> Hmm, but you get the attitude from 3? Do you also get the attitude from 1? This is confusing ...

No, I can not get attitude from 1, when I try get attitude from 1 with get_attitude, the result is all Nan

julianoes, 2025-07-25:
Ok, I'm just trying to reproduce this. I'm using gimbal version 0.3.8 at the moment.

I can control the gimbal from the PX4 MAVLink shell, using e.g. `gimbal test pitch -45`.

The gimbal example fails with:

```
cmake --build build && build/gimbal/gimbal serial:///dev/serial/by-id/usb-Auterion_PX4_FMU_v6C.x_0-if00:921600
ninja: no work to do.
[08:18:31|Info ] MAVSDK version: v3.8.0 (mavsdk_impl.cpp:32)
[08:18:31|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:950)
[08:18:31|Debug] Component Autopilot (1) added. (system_impl.cpp:473)
[08:18:31|Debug] Component Gimbal (154) added. (system_impl.cpp:473)
[08:18:31|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:313)
[08:18:31|Debug] Discovered 2 component(s) (system_impl.cpp:591)
[08:18:31|Debug] MAVLink: critical: Preflight Fail: Yaw estimate error (system_impl.cpp:339)
[08:18:32|Debug] MAVLink: info: GCS connection regained	 (system_impl.cpp:339)
[08:18:33|Warn ] sending again after 0.50008 s, retries to do: 1  (512). (mavlink_command_sender.cpp:336)
[08:18:33|Warn ] Request was for msg ID: 283 (mavlink_command_sender.cpp:342)
[08:18:33|Warn ] Retrying failed for REQUEST_MESSAGE command for message: 283, to (1/154) (mavlink_command_sender.cpp:375)
[08:18:34|Warn ] sending again after 0.501303 s, retries to do: 1  (512). (mavlink_command_sender.cpp:336)
[08:18:34|Warn ] Request was for msg ID: 283 (mavlink_command_sender.cpp:342)
Could not find a gimbal, exiting...
```

Message 283 is [GIMBAL_DEVICE_INFORMATION](https://mavlink.io/en/messages/common.html#GIMBAL_DEVICE_INFORMATION).

Usually I would have expected it to work even without GIMBAL_DEVICE_INFORMATION available, but here we are...

Let me try to upgrade to v0.3.9 next.

julianoes, 2025-07-25:
I just found this related thread which looks similar:
https://discuss.px4.io/t/siyi-a8-mini-with-mavsdk/40439

julianoes, 2025-07-25:
I don't see anything from gimbal ID 3, not sure how I should be reproducing it.

jdx200512, 2025-07-25:
> How did you get that to work?

I can get gimbal_device_information

[image]

does there other param that affect?

julianoes, 2025-07-29:
I've figured out why the example didn't work, see #2629.

At this stage, I don't know how to reproduce your problem. Please let me know what else I should try.