mavlink/MAVSDK #2639: Bus error on Raspberry Pi with 3.9.0

tsantzi, 2025-08-12:
As per the title, I get a bus error on a Raspberry Pi armhf. Installed via release .deb.

Running under gdb gives:

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[02:48:16|Info ] MAVSDK version: v3.9.0 (mavsdk_impl.cpp:32)
[New Thread 0xf7146400 (LWP 3085)]
[New Thread 0xf6945400 (LWP 3086)]
Connecting...
[02:48:16|Warn ] Connection using tcp:// is deprecated, please use tcpin:// or tcpout:// (cli_arg.cpp:41)
[New Thread 0xf5fff400 (LWP 3087)]
Waiting for system...
[02:48:16|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:953)
[New Thread 0xf55ff400 (LWP 3088)]
[New Thread 0xf4dfe400 (LWP 3089)]
[02:48:16|Debug] Component Autopilot (1) added. (system_impl.cpp:473)

Thread 4 "main" received signal SIGBUS, Bus error.
[Switching to Thread 0xf5fff400 (LWP 3087)]
0xf7acbd54 in float mav::Message::_readSingle<float>(mav::Field const&, int) const () from /lib/libmavsdk.so.3
```

Works fine after downgrading to 3.8.1 (and 2.13.0 that I was using before that)

julianoes, 2025-08-12:
Can you get me the backtrace at that moment?

tsantzi, 2025-08-13:
Sure, here you go:

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[07:46:04|Info ] MAVSDK version: v3.9.0 (mavsdk_impl.cpp:32)
[New Thread 0xf7146400 (LWP 22270)]
[New Thread 0xf6945400 (LWP 22271)]
Connecting...
[07:46:05|Warn ] Connection using tcp:// is deprecated, please use tcpin:// or tcpout:// (cli_arg.cpp:41)
[New Thread 0xf5fff400 (LWP 22272)]
[07:46:05|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:953)
[New Thread 0xf55ff400 (LWP 22273)]
[New Thread 0xf4dfe400 (LWP 22274)]
[07:46:05|Debug] Component Autopilot (1) added. (system_impl.cpp:473)
Waiting for system...

Thread 4 "main" received signal SIGBUS, Bus error.
[Switching to Thread 0xf5fff400 (LWP 22272)]
0xf7acbd54 in float mav::Message::_readSingle<float>(mav::Field const&, int) const () from /lib/libmavsdk.so.3
(gdb) bt
#0  0xf7acbd54 in float mav::Message::_readSingle<float>(mav::Field const&, int) const () from /lib/libmavsdk.so.3
#1  0xf7ac8604 in mav::Message::getAsNativeTypeInVariant(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const () from /lib/libmavsdk.so.3
#2  0xf7801d44 in mavsdk::LibmavReceiver::libmav_message_to_json[abi:cxx11](mav::Message const&) const () from /lib/libmavsdk.so.3
#3  0xf57ff000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

julianoes, 2025-08-13:
I tried on my RPi 5 with `arm64` but couldn't reproduce it yet. What RPi is this on, and what arch? You mention `armhf` so that would hint to an older 32bit one?

julianoes, 2025-08-13:
Another thought, so far I had been compiling on the RPi and not used the library files which are cross compiled in dockcross, so that might also make a difference.

julianoes, 2025-08-13:
Ok,  I can reproduce the issue!

julianoes, 2025-08-13:
This will hopefully fix it: https://github.com/julianoes/libmavlike/pull/1

julianoes, 2025-08-13:
Fix https://github.com/mavlink/MAVSDK/pull/2641 and release coming soon.

closed 2025-08-14 by julianoes