mavlink/MAVSDK #2642: How to send statustext message?

tranhien1612, 2025-08-14:
I tried mavsdk lib to connect to mavlink and [ardupilot](https://github.com/ArduPilot/ardupilot) to simulate the plane and use qgc to display.
I have connected and sent control commands. I want to send statustext command to display information on `Vehicle Messages` tab on qgc. 
i used the code below to do it, it doesn't give any error but also doesn't display.

```c
std::shared_ptr<mavsdk::System> vehicle;

mavsdk::Mavsdk mavsdk{
    mavsdk::Mavsdk::Configuration{mavsdk::ComponentType::GroundStation}
};
vehicle = mavsdk.systems().at(0);

...
...

void send_qgc_message(const std::string& text) {
    auto passthrough = mavsdk::MavlinkPassthrough{vehicle};
    mavlink_message_t message;
    mavlink_statustext_t statustext;

    statustext.severity = MAV_SEVERITY_WARNING; // MAV_SEVERITY_INFO, WARNING, etc.
    std::strncpy(statustext.text, text.c_str(), sizeof(statustext.text) - 1);
    statustext.text[sizeof(statustext.text) - 1] = '\0';

    mavlink_msg_statustext_encode(
        1,
        1,
        &message,
        &statustext
    );
    auto ret = passthrough.send_message(message);
    if (ret != mavsdk::MavlinkPassthrough::Result::Success) {
        logger.log("Failed to send message");
    }
}
```

[image]

pls help me to do it. Thanks

julianoes, 2025-08-14:
The message is sent from you as a ground station because that's your configuration.

You should then also use `queue_message` instead.

tranhien1612, 2025-08-14:
I am not familiar with this, do you have an example for it?

julianoes, 2025-08-14:
It might only show them coming from 1/1, so you have to fake this:
```
            mavlink_address.system_id,
            mavlink_address.component_id,
```

with 
```
1,
1
```

instead.

Oh, and then there is the problem that both MAVSDK and MissionPlanner connect to the drone, right? So you're assuming the drone would forward your message but if the message's source is "from" the drone, it might not do that. You will have to think of how the communication path should work.

You could for instance connect to the drone and MissionPlanner in MAVSDK and forward messages between them.

See: https://mavsdk.mavlink.io/main/en/cpp/guide/connections.html#forwarding-mavlink-between-connections