mavlink/MAVSDK #2652: Does Mavsdk API support Ardupilot firmware ?
Or is it only compatible with PX4 firmware ?

If yes , please tell me the procedure to connect to the drone through udpout through Mavsdk from my laptop

riteshrkulkarni, 2025-09-13:


julianoes, 2025-09-13:
In general yes. With specific functionality there might be holes that need fixing.

It seems like you can't connect, so then the question is how the Pixhawk is connected to your laptop and what settings you tried with MAVSDK.

ytyisme, 2025-09-24:
> In general yes. With specific functionality there might be holes that need fixing.
> 
> It seems like you can't connect, so then the question is how the Pixhawk is connected to your laptop and what settings you tried with MAVSDK.
Need a list showing which features still have issues in ArduPilot.

ZhongmouLi, 2025-10-24:
@ytyisme That’s a great idea! Do you know if the development team has any plans for this?

I’d be happy to contribute to supporting ArduPilot if there’s a list of desired features or tasks.

julianoes, 2026-09-15:
Duplicate of https://github.com/mavlink/MAVSDK/issues/1568

closed 2026-09-15 by julianoes