mavlink/MAVSDK #2653: MavlinkPassthrough unable to receive DATA32 messages

nocs85, 2025-09-13:
Hi all,

I'm having trouble using `mavsdk::MavlinkPassthrough` to receive `DATA32` MAVLink messages (Message ID = 170).

**Setup**

- **Autopilot**: Onboard, running ArduPilot
- **Companion Computer**: Sends `DATA32` messages, uses **component ID 192**

**Problem**

The Companion sends `DATA32` messages, and I'm trying to catch them using the following code:

```
mavsdk::MavlinkPassthrough m_passthrough;
m_passthrough.subscribe_message(170, [](const mavlink_message_t& message) {
          mavlink_data32_t data_msg;
          mavlink_msg_data32_decode(&message, &data_msg);

          std::cout << "Received DATA32 message " << std::endl;
}
```

However, the callback is **never triggered**.

**Debugging Attempts**

Initially, I suspected the issue might be related to the **component ID** (Companion uses 192). To test this, I subscribed to the `HEARTBEAT` message, and that **does work**:

```
m_passthrough.subscribe_message(MAVLINK_MSG_ID_HEARTBEAT, [](const mavlink_message_t& message) {
          mavlink_heartbeat_t heartbeat;
          mavlink_msg_heartbeat_decode(&message, &heartbeat);

          uint8_t sysid = message.sysid;   // System ID of the sender
          uint8_t compid = message.compid; // Component ID of the sender
          std::cout << "HEARTBEAT from sysid: " << static_cast<int>(sysid) << ", compid: " << static_cast<int>(compid) << "\n";
}
```

So I know messages from the Companion are being received.

**Validation from Python**

To confirm that the `DATA32` messages are being sent correctly, I used the following Python script (via `pymavlink`), and it does receive the messages:

```
from pymavlink import mavutil

# Create a connection listening on UDP port 14551
master = mavutil.mavlink_connection('udpin:0.0.0.0:14551')

print("Waiting for messages...")

while True:
    # Wait for the next message with a timeout of 1 second
    msg = master.recv_match(type=None, blocking=True, timeout=1)

    if msg is None:
        continue  # Timeout, no message received

    # Check if the message ID matches 170
    if msg.get_msgId() == 170:
        print("received")
```

This confirms that `DATA32` messages are sent and received with the correct payload.

I'm currently stuck. My gut feeling is that the issue might be related to `DATA32` not being part of the standard MAVLink protocol, but instead part of the ardupilotmega dialect.

I’ve searched around but haven’t found a definitive answer. So my questions are:

- Is t his a bug or known limitation in MavSDK?
- Does MavSDK's MavlinkPassthrough support custom or dialect-specific messages like `DATA32`?
- Is there a workaround or known approach to handle these messages correctly?

I’d really appreciate any help or guidance—I'm out of ideas at this point.

Thanks in advance!

julianoes, 2025-09-14:
> My gut feeling is that the issue might be related to DATA32 not being part of the standard MAVLink protocol, but instead part of the ardupilotmega dialect.

That's exactly it.

I'm meaning to switch to [all.xml](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/all.xml) which includes common.xml as well as ardupilotmega.xml but I haven't done so yet.

For now, you need to switch to the new [MavlinkDirect plugin](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mavlink_direct/include/plugins/mavlink_direct/mavlink_direct.h) and load the xml for that message:

```
mavlink_direct.load_custom_xml(R"(
    <message id="170" name="DATA32">
      <description>Data packet, size 32.</description>
      <field type="uint8_t" name="type">Data type.</field>
      <field type="uint8_t" name="len" units="bytes">Data length.</field>
      <field type="uint8_t[32]" name="data">Raw data.</field>
    </message>
)");

    // Subscribe to message
    auto handle = mavlink_direct.subscribe_message(
        "DATA32", [](const mavsdk::MavlinkDirect::MavlinkMessage& message) {
            std::cout << "Got message DATA32 with fields" << message.fields_json << std::endl;
        });
```

I hope I have the escaping correct, but you get the idea.

julianoes, 2025-09-14:
It is a valid alternative, however, I plan to deprecate mavlink_passthrough with v4 eventually because I want to get away from installing/providing the MAVLink C includes with MAVSDK. It will still be a compile-time option though if you still need it, just not included by default.

closed 2025-09-14 by julianoes