mavlink/MAVSDK #2656: GPS Info Not Updating Correctly: Satellites and FixType Always Return Defaults in MAVSDK

lukasrml, 2025-09-17:
Hello, I was trying to use the _telemetry->gps_info() command in my C++ code to print the number of satellites found by my drone and the fix type recognized by it. So I used the following code:

std::cout
  << "NumSat=" << _telemetry->gps_info().num_satellites
  << "FixType=" << _telemetry->gps_info().fix_type

Everything else in the code works fine, and it connects perfectly to the drone simulation in Gazebo, but when it prints, it always returns the following results:

NumSat=0
FixType=No GPS

This is not what I expected, because when I connect the same simulation to QGroundControl, I can see that there are 10 satellites connected and FixType is set to Fix3D Lock. Additionally, when I run the code connected to my drone, it continues to show the same error, not displaying the expected number of satellites or FixType.

What is the correct way to use these commands? How can I solve this problem?

julianoes, 2025-09-17:
That's odd.

The Mavlink message is processed here:

https://github.com/mavlink/MAVSDK/blob/662b9e9db09b2dacca376a2e567893cffc42779c/src/mavsdk/plugins/telemetry/telemetry_impl.cpp#L926-L990

Can you double check whether `GPS_RAW_INT` is being sent? You can do that in QGC Mavlink Inspector or using the sniffer example (https://github.com/mavlink/MAVSDK/tree/main/examples/sniffer).

julianoes, 2025-09-22:
I'm not sure what's going on. Can you make sure you are on top of latest MAVSDK v3.10.2?